LaserWeb/deprecated-LaserWeb3 #1: Best Practise:  Smoothie Buffer

openhardwarecoza, 2016-06-02:
@Wolfmanjm

Right, if I need to write a whole new buffer for SPJS, i'd rather start over fresh (In nodejs, where i am comfortable, and without the few hundred thousand lines of SPJS code to confuse me)

So, what I built here:  https://github.com/openhardwarecoza/LaserWeb3/blob/master/server.js
node-serialport:  Connect to a serail endpoint.  

I have an array called gcodeQueue

To this queue I can prepend data (to push a request for position feedback into the queue in front) or append to the back - with respectively

https://github.com/openhardwarecoza/LaserWeb3/blob/master/server.js#L151 - jumpQ (puts it in front)
https://github.com/openhardwarecoza/LaserWeb3/blob/master/server.js#L147 - addQ (adds it last)

Next I can take entries from this array, by shifting them out into a var https://github.com/openhardwarecoza/LaserWeb3/blob/master/server.js#L161-L164 and writing it to the serial port.

the send1Q function only sends one LINE of gcode.

It gets called after a 50ms delay after an OK (so simple ping pong buffer)

Now, theroetically, for marlin, that should result in a send, ok, send, ok. 

But your conversation with regards to the USB driver buffer, has me doubting my own plans so just want to clarify if the above should work?

wolfmanjm, 2016-06-02:
No idea if this works for Marlin or not... I suspect not too well. 
However for Smoothie it should work fine, although I'd get rid of the 50ms delay, when you get an ok you can send the next line immediately.
Note that it is one ok per line, so if the line has multiple G codes you will still only get one G code (this can be changed with a config entry though if you prefer one ok per gcode).

(AFAIK for Marlin you need to fill its serial buffer to avoid jerkiness, but you should check with the Marlin developers, same with GRBL).

openhardwarecoza, 2016-06-02:
I have a seperate issue here (#2) for Marlin,  to get the right advice
directly from them.

I am breaking \n and sending each on its own so i can get an ok per line.

Right,  so if your happy with it,  i'll continue and add some error handling

wolfmanjm, 2016-06-02:
Also note that Smoothie responds to ? the same as GRBL does, and there is no \n needed for the ? otherwise you get an ok for the blank line.
So your test for GRBL by sending ? will confuse things as you may think it is grbl not smoothie.

openhardwarecoza, 2016-06-02:
All cool,  i am not going to use grbl mode at all here.   I am writing a
Smoothie buffer,  a Marlin buffer,  a Grbl buffer all on their own,  user
can pick.  Avoid the jack of all trades master of none approach

wolfmanjm, 2016-06-02:
You should use ? and not M114 to get current position. They are totally different.
M114 is not immediate, and returns the last received position based on G0/G1 and not the current position. 
? returns the realtime position and returns immediately (no ok).

openhardwarecoza, 2016-06-02:
Ahhh!  Thats good to know then!  -ok will grab  ? then

wolfmanjm, 2016-06-02:
you said "Now, theroetically, for marlin, that should result in a send, ok, send, ok." that is why I mentioned Marlin, did you mean Smoothie? in which case send, ok, send works fine.

openhardwarecoza, 2016-06-02:
Yip,  smoothie.  Title of this issue is smoothie.  #2  is marlin,  #3 is
grbl.    Based on the spjs fight on LaserWeb2,  this is now LaserWeb3 where
i am trying to trash spjs and make our own interface so we have control.
Therefore the three issues,  to ask each relevabt firmware master,  how
best to address their firmware.  No finger pointing (;

openhardwarecoza, 2016-06-03:
@wolfmanjm is there a debug functionality to ask Smoothie to echo what it received?  (Marlin and Grbl has - makes it easy while testing queuing code)

wolfmanjm, 2016-06-03:
no sorry there is not, not really sure how that would help either.

openhardwarecoza, 2016-06-03:
Right, figured out my issue why I wanted debug.  I actually was sending before the OK (oops) 
Fixed and as of this version https://github.com/openhardwarecoza/LaserWeb3/blob/2cdf067659989c6dbb136bb2b8cda7fda19fc410/server.js

I have reliable lineup between emptying the queue, and counting the OKs coming back (:

I just need to add some error handling (Smoothie returns 'rs' if it needs a resend?  https://github.com/Smoothieware/Smoothieware/blob/100e5055156f7fbe9f7b57fccdc4bfd0784bc728/src/modules/communication/GcodeDispatch.cpp#L425-L426

Do you have a shortlist of other responses I need to look for?

```
ok  = ok
<  = response to ? 
rs = resend

```

As a sidenote, see why Smoothie is my favourite firmware, with your help in less than a day, we are done.  Grbl, Sonny gave some info so i can start that next, Marlin's devs hasnt even bothered answering lol.  Thanks for the support Jim! It really is appreciated

wolfmanjm, 2016-06-03:
we mostly follow the reprap spec found here...  http://reprap.org/wiki/G-code#Replies_from_the_RepRap_machine_to_the_host_computer

You can probably ignore the RS because you should never see it as USB has a checksum itself, ditto with linenumbers you can probably ditch them and the checksum.
There is a protocol it would be nice if you followed that allows suspend and resume commands to be sent to the host, also documented above... ```// action:suspend``` etc

openhardwarecoza, 2016-06-06:
Ok, so waiting for OK has slowed us down.  I think before, when we weren't waiting for the OK, the USB buffer prevented any disasters, but also fed the queue quick enough...

Waiting for OK makes the planner empty (since theres only one move at a time?) so consecutive moves have decelerate, accelerate in between them (not great for laser rastering)

wolfmanjm, 2016-06-06:
No that is not how it works. it sends OK when the command is received, not when it is executed so it won't slow anything down, all the other hosts do it this way and it works fine. So long as you feed it as fast as you can with no delay (you said you had a delay in there). And you need to feed it faster than the move can execute, so if it is a very short move then the planner never gets to see more than one move a t a time. if it is a move greater than about 50ms then the planner queue will fill up.
FYI there is no USB buffer. it doesn't work that way. But USB does have its own flow control, and that is why you would get deadlock if you try to send lines before you get an OK, unless reading the OK is in a separate thread, so one thread to send one thread to receive.

wolfmanjm, 2016-06-06:
Now if the moves you are sending are extremely short then yes you will have an issue, but it has nothing to do with the USB speed or waiting for OK. You simply have a convoy effect, you are consuming faster than you can provide. There is no fix for that I am afraid, even if you were able to fill the planner queue before it started executing, it would still run dry after 32 moves. and then start doing one move at a time. This is standard communications theory.

openhardwarecoza, 2016-06-06:
Thanks @wolfmanjm and @HakanBastedt - both reminded me of delay.

I took it out, now just waiting on @cojarbi to test and le me know

openhardwarecoza, 2016-06-09:
@wolfmanjm , i dont seem to be able to get a response from "?" over ethernet? 

```
queryLoop = setInterval(function(){ runCommand('?') }, 300);

function runCommand(cmd, silent) {
  var host = $('#smoothieIp').val();
    cmd += "\n";
    url = silent ? "http://" + host + "/command_silent" : "http://" + host + "/command";
    // Send the data using post
    var posting = $.post(url, cmd);
    // Get the result
    if (!silent) {
        posting.done(function(data) {
            $.each(data.split('\n'), function(index) {
                console.log(this)
            });
        });
    }
}
```

I do get response to M114, but not to "?"

wolfmanjm, 2016-06-09:
? is only supported on the UART and USB serial ports. There is no way to do ? on the ethernet as an immediate command. I think get status will work on ethernet, although it is not immediate, but will return pretty fast.

openhardwarecoza, 2016-06-09:
Thanks @wolfmanjm !  'get status' did the trick! Since this isnt as live as websockets, I anyway only poll every 300ms so "pretty fast" is fast enough (:

openhardwarecoza, 2016-07-26:
Right,  i am happy with the usb buffer, lots of happy users too.  Closing this issue.  I have smoothie ethernet on a seperate ticket

closed 2016-07-26 by openhardwarecoza