LaserWeb/deprecated-LaserWeb3 #2: Best Practise: Marlin Buffer

openhardwarecoza, 2016-06-02:
@thinkyhead @nextime @HakanBastedt

Right, if I need to write a whole new buffer for SPJS, i'd rather start over fresh (In nodejs, where i am comfortable, and without the few hundred thousand lines of SPJS code to confuse me)

So, what I built here: https://github.com/openhardwarecoza/LaserWeb3/blob/master/server.js
node-serialport: Connect to a serial endpoint.  Keeping it VERY simple to start off...

I have an array called gcodeQueue

To this queue I can prepend data (to push a request for position feedback into the queue in front) or append to the back - with respectively

https://github.com/openhardwarecoza/LaserWeb3/blob/master/server.js#L162- jumpQ (puts it in front)
https://github.com/openhardwarecoza/LaserWeb3/blob/master/server.js#L158 - addQ (adds it last)

Next I can take entries from this array, by shifting them out into a var https://github.com/openhardwarecoza/LaserWeb3/blob/master/server.js#L170-L177 and writing it to the serial port.

the send1Q function only sends one LINE of gcode.

It gets called after a 50ms delay after an OK (so simple ping pong buffer) - I also manually call it after adding a M114 to the queue: https://github.com/openhardwarecoza/LaserWeb3/blob/master/server.js#L114-L115

Now, theoretically, for marlin, that should result in a send, ok, send, ok.?

HakanBastedt, 2016-06-02:
It is of course best if @thinkyhead chime in with an answer.
My understanding is that Marlin responds with "ok" when the command has been received and has been processed (data put in p-buffer and whatever). There may be another response "Unknown command:", "resend:", but let's skip that for the time being.
Ping-pong with send ok send ok is fine. Four lines can be buffered so send send send send ok send ok send ok ok ok ok is also fine. sends - oks < 4 though.

openhardwarecoza, 2016-06-02:
See thats what i wasnt sure of,  should we prepend?  Laser moves are short
so we dont want to starve the buffer either.  @thinkyhead ?

And yes,  i havent added any error checking,  building up from nothing this
time so i dont have to reverse engineer anyone elses code (;

HakanBastedt, 2016-06-02:
I suddenly felt I wasn't sure how this "4" works.
I can try the code though. See if I can figure out to "run" it.

openhardwarecoza, 2016-06-02:
Install instructions:

Install nodejs
Git pull laserweb3
Cd into laserweb3
npm install
npm install serialport@4.0.0beta3
node server.js
Open browser to 127.0.0.1:8000   (same as LaserWeb1 was but bumping up
version of node serial to 4.0.0beta3 latest as of today)

HakanBastedt, 2016-06-02:
Got it to build and start. Magic things happens when accessing http:/xx:8000 though,
LaserWeb2 actually starts. Should it? Expected interface to server.js.
Tried to connect to server.js instead of to spjs, error is
`ERROR:  fileServer error:/ws :  Not Found`

openhardwarecoza, 2016-06-02:
Yip,  thats a whole different Laserweb that you see on :8000.   The one at
http://openhardwarecoza.github.io/LaserWeb2 wont be able to talk to the
Nodejs backend.  (;

Thats why this one is called LaserWeb3 -  and while i'm doing that it also
makes LaserWeb offline available again (was like the biggest user complaint
nezt to. Spjs,  with laserweb1 we hosted it locally,  LaserWeb2 went
cloud,  now in 3, back to local

HakanBastedt, 2016-06-02:
Ah I see. I will test it.
But first I will record the serial traffic between a well known, reliable host and the Marlin laser board. See if I understand the "4" in reality.

openhardwarecoza, 2016-06-02:
Good idea!   What logic analyser do you have,  sounds like something i need
to put on my wishlist

HakanBastedt, 2016-06-02:
This is the logic analyzer I have, software to download from Saleae.
http://www.ebay.com/itm/USB-Logic-Analyzer-Device-Set-USB-Cable-24MHz-8CH-24MHz-for-ARM-FPGA-M100-/182121929594?hash=item2a67501b7a:g:qb4AAOSwt7pXLUOv First occasion I use it and I like it. I record 2 minutes of data, ask the software to decode it to serial data and then sort the lines out in an editor.
The "4" means sends-oks <= 4. So four more sends than oks is allowed. It even says five, but the ok and next send cross each other so 4.

HakanBastedt, 2016-06-02:
Ok I tested the server.js and it works in the meaning that it connects and commands are executed. But the flow control isn't there, there are these occasional moves- reminds me of the first version of spjs. Also, the DRO is not updated, but the correct values are shown as output from server.js. 
I think this is it for today.

openhardwarecoza, 2016-06-03:
Thanks for the tests

DRO: Not added yet at all.
I only have Smoothie's left on my desk - so therefore server.js and https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/js/socketcomms.js has basic functions to parse smoothie (I actually ran a job yesterday using it and for Smoothie we are on the momey)

For marlin and grbl I havent added anything
I first wanted to get feedback from the @MarlinFirmware @thinkyhead team on how it SHOULD be (As an engineer I like the planned approach a little more than trial and error.   Smoothie and Grbl already got back to me in #1 and #3 and therefore I can continue.  Also. they already have their protocols nicely documented.  With Marlin, we are in the dark a bit :(

Basically, in the first week or two of LaserWeb3 - create a fork and make it work with marlin. Dont worry about pull request compatibility.  Once it works solid for each of the three major firmwares, only then will I spend the time to work them into a single instance (probably by making socketcomms.js and server.js modules for each firmware)

HakanBastedt, 2016-06-03:
I definitely see your point about info from the right source. @AnHardt maybe can share info on how the client is supposed to communicate with Marlin and how to use the BUFSIZE command lines?

I can test with the Smoothie uuhm buffer is no the right word anymore, functions. Afaik Marlin should work with that also. Next step would be to allow 4 extra sends, and that's it basically.

openhardwarecoza, 2016-06-03:
Well, while Marlin devs continue to ignore us (;  with the help of the Smoothie developers, the smoothie buffer is solid now 
https://github.com/openhardwarecoza/LaserWeb3/issues/1#issuecomment-223520643

AnHardt, 2016-06-03:
Pingpong is save for Marlin. Send a command, wait for an 'ok', send the next command, wait for an 'ok', ....

We have 4 command-line buffers but there is no feedback about how many of them are used. (except you compiled in ADVANCED_OK. For now lets ignore that)    

We do not have BUFSIZE command lines. We do have BUFSIZE  defines in our configuration files, So we can adjust the RX buffer size at compile time - not on the fly. Standard size is 128 byte. Don't trust on this size but, make it configurable in your application.    

If you want to speed up the communication with Marlin you can use the RX-buffer. You can send that many lines of g-code until the RX-buffer is full. There is no feedback how full the buffer is. You have to count. That's the same what the better GRBL-senders did two years ago. (I have not looked in their code since then.)
I'd do that with a ring buffer where i'd spore the previous command length.
1. Send a line.
2. Store it's length in the new ring buffer.
3. If you received an ok delete the oldest entry in the ring buffer.
4. Sum up the ring buffer content.
5. If there is enough space for your next command line go to 1.
6. Else go to 3.

thinkyhead, 2016-06-04:
> while Marlin devs continue to ignore us

Sorry, I have a lot of balls in the air. Don't take it personally.

HakanBastedt, 2016-06-04:
That is indeed so. Learned something new again, that is the great think with all this.
In that case, I guess the right thing to do is to go with the grbl way.
I'll have a look in the new server-marlin and see if I can get it to work.

openhardwarecoza, 2016-06-04:
@thinkyhead no problem, and it was a very joking casual comment -  (compared to how fast Smoothie and Grbl got back in #1 and #3 

I read about your sabotage - so understandably you guys are very busy!  Good luck! And thanks for the awesome reply! Ring buffer it is then (Much more clear than the community recommended "send 4 lines and count OKs"  - thats why i opened these issues, and tagged the actual developers.   Best practise comes from the guy who Knows his sh*t (;  - not the end user who makes assumptions.  

Now we know, we can implement it properly

thinkyhead, 2016-06-04:
> …your sabotage…

Just one? I feel like one nearly occurs every few months. 😄

openhardwarecoza, 2016-06-04:
Wow, I left the marlin eco system years ago when I closed my 3dprinter store, so didnt know about that.  Only saw the recent one when you added me to the Hosts developers team! Sorry, that must be demotivating.

Just always remember, people are assholes, fire them if their personalities suck, and join a "Open source developers support group" where we can all vent about dealing with people and "contributors with baggage"

PS: You may mute of this, otherwise the notifications will drive you crazy. Me an Hakkan will be chatting in thread here to start gettng this under way

PS:  Go checkout openhardwarecoza.github.io/LaserWeb2 (;  if one of your users come looking for a good Laser cutter host - send them our way (;

openhardwarecoza, 2016-06-04:
@HakanBastedt - I pushed my latest code for Smoothie into server-smoothie.js

I'd say copy that over into server-marlin.js and then start changing (so you have the latest socket.emits for the UI)

HakanBastedt, 2016-06-04:
After endless tests and not understanding why it doesn't work when it ought I finally figured it out.
Both the input and output from serial port has to be **unbuffered**, especially the input.

I programmed the writing with queues and all in C++ (which I know bit better), got the same behavior as in server-marlin.js when is *should* work. Nailed it down to the internal buffering in ifstream and ofstream. Switch them of and it started to work just fine.

openhardwarecoza, 2016-06-05:
port.flush(function(err,results){}); should clear the 'buffer' in node serialport,  maybe we should flush after every write -  but we can also discuss this with @EmergingTechnologyAdvisors    https://github.com/EmergingTechnologyAdvisors/node-serialport

openhardwarecoza, 2016-06-05:
https://github.com/EmergingTechnologyAdvisors/node-serialport/blob/7e20789bf0ef5743448b2b24058311f741218b66/README.md#write-buffer-callback

HakanBastedt, 2016-06-05:
Thanks for the links. I'll have to make it work in sever-marlin.js now. I have modded Marlin to report characters in Rx buffer and there is a big difference with js compared to my c++ code and Repetier-host. server-marlin doesn't fill up the buffer as "it should", so that is what I am after. It may take a day or two I guess.

nextime, 2016-06-05:
@HakanBastedt can i suggest to change point of view? ( not that this make a big difference ).

server-marling doesn't need to fill up the buffer, it needs to assure it never get empty waiting for new commend without having nothing to do.

HakanBastedt, 2016-06-05:
That is true of course. All suggestions are welcome. I use the fill level of the buffer as a diagnostic tool.  I have a well working client which is repetier-host. I see from the logic analyzer how it does it. When I switched off the buffers, my c++ programs does it the same way and is rock solid as well. server-marlin.js does it differently, even it it should be the same. And after 5000 lines there is now "echo:unknown command" with missing characters. Same as I got with c++ with buffering.
Anyway I am on the problem, but there is also quite nice weather outside :)

openhardwarecoza, 2016-06-06:
FYI - adding the ping pong buffer to Smoothieware has taken us back aa step on the smoothie side.  Now we only have one move at a time in the buffer so the underruns makes the raster moves stop and go between segments ): 

Hosts are hard

openhardwarecoza, 2016-06-06:
On a sidenote @HakanBastedt , if we want to use C and can make it cross platform, its worth investigating.  A little C app compiled - is easy to distribute, getting users to install a nodejs stack isnt the best thing in the world anyway

HakanBastedt, 2016-06-06:
I am still doing stupid mistakes in the coding, so won't bother other people just yet. There is still hope that it will work in js. Errors show up after like 5000-10000 lines of G-code so there is plenty of time to do other things in between. 

I thought Smoothie's buffer size was "large". If you have the code with settimeout, I would try and remove that delay there is no reason to wait with a "real" usb connection, afaik.

Another option is to go back to spjs and try to sort out the out of band data.

HakanBastedt, 2016-06-07:
Ping-pong works for Marlin. But I can't get any other buffer-filling tactics to work from js. Sooner or later there is data corruption "error: Unknown command..." or the "ok" just stops coming back. The problem can occur anywhere after 100 lines or 24000 lines. The Marlin board? When in doubt I run my c++ program and it feeds the 25000 lines to the board just fine. It uses ping-pong with an extra G-code inserted if size allows, at regular intervals. Keeps the buffer nicely full, In the beginning there is some time needed to fill up the buffer, but then it is mostly a matter of ping-pong with the buffer almost full.

I read a bit of grbl serial issues, and it seems to be related.

openhardwarecoza, 2016-06-11:
https://github.com/w-A-L-L-e/printerface/blob/master/printerface.js how about spawning printrun's pronsole and using stdin/stdout  going to be a less easy install,  but low hanging  fruit to get all these marlin users off my case (;

openhardwarecoza, 2016-06-12:
@HakanBastedt - have you tried https://github.com/andrewhodel/reprapweb by the way?  Also a nodejs backend.

HakanBastedt, 2016-06-12:
I have not tested them, but both are worthwhile to test. 
The first one should be a safe bet - I assume :)
The second one seems nice, he is using serialport if I see right?

I'll see if I can test both later tonight.

I don't know how to go further with server-marlin.js, everything I have tried has failed so far except ping-pong. It's in hold for me.

openhardwarecoza, 2016-06-12:
> The second one has some legacy from laserweb1 days,  i had a lot of co
> working with him but in the end the relationship went sour.    That said,
> it should be a very easy,  quick test.  If that works reliable,  its a
> simple ping pong buffer, using nodejs serialport.

Oh and ps,  if ping pong is stable,  its better than what we currently have
(aka no marlin support lol)

HakanBastedt, 2016-06-12:
Got reprapweb to work and yes it works with my Marlin. At least the 25000 lines G-code file was ok,
But in that case I think it I put back the pingpong in server-marlin.js and than you can use that.

HakanBastedt, 2016-06-13:
Just pushed a modified server-marlin.js to a new branch marlin-comm.
It works with ping-pong, passed my 25 klines g-code test file, so at least it is something.
It looks like everything has changed but a lot is my editor's indentation.
Neat with the moving cross-hair btw.

openhardwarecoza, 2016-07-27:
So I'm almost at the Marlin point - Grbl and Smoothie are both working nicely.

openhardwarecoza, 2016-09-02:
Had a user report success using https://github.com/openhardwarecoza/LaserWeb3/blob/marlin-comm/server-marlin.js  

See https://plus.google.com/102797371787170159448/posts/2Dzt8WX6hT8

This was with https://github.com/Downunder35m/MarlinKimbra-K40

openhardwarecoza, 2016-10-23:
Moved to wishlist on the readme https://github.com/openhardwarecoza/LaserWeb3/commit/b39c56a24a7de76afdbf95dd3e28ae12f6bf6c8c so I can close the issue for now

closed 2016-10-23 by openhardwarecoza