LaserWeb/deprecated-LaserWeb3 #3: Best Practise: Grbl Buffer

openhardwarecoza, 2016-06-02:
@chamnit 

Right, if I need to write a whole new buffer for SPJS (refer the fight in issue 52 on openhardwarecoza/LaserWeb2), i'd rather start over fresh (In nodejs, where i am comfortable, and without the few hundred thousand lines of SPJS code to confuse me)

So, what I built here: https://github.com/openhardwarecoza/LaserWeb3/blob/master/server.js
node-serialport: Connect to a serial endpoint. Keeping it VERY simple to start off...

I have an array called gcodeQueue

To this queue I can prepend data (to push a request for position feedback into the queue in front) or append to the back - with respectively

https://github.com/openhardwarecoza/LaserWeb3/blob/master/server.js#L162- jumpQ (puts it in front)
https://github.com/openhardwarecoza/LaserWeb3/blob/master/server.js#L158 - addQ (adds it last)

Next I can take entries from this array, by shifting them out into a var https://github.com/openhardwarecoza/LaserWeb3/blob/master/server.js#L170-L177 and writing it to the serial port.

the send1Q function only sends one LINE of gcode.

It gets called after a 50ms delay after an OK (so simple ping pong buffer) - I also manually call it after adding a M114 to the queue: https://github.com/openhardwarecoza/LaserWeb3/blob/master/server.js#L114-L115

Now, theoretically, for marlin, that should result in a send, ok, send, ok.

Can you advise me, what is the best way to get this done to Grbl?
We never could get it working nicely in Laserweb v 1  -  https://github.com/openhardwarecoza/LaserWeb/issues/94

In LaserWeb v2 it works, but depends on Serial Port JSON server (and I'm tired of being told of TinyG's superiority - so i'm moving on) 

Appreciate any help, lets hope 3rd times the charm and we get Grbl + laserWeb playing really nicely under node-serialport!

chamnit, 2016-06-02:
@openhardwarecoza : I'm not familiar with Node or Go, so I don't think I can be much help there. But, the streaming protocol is outlined in Grbl's Interface Wiki page. At its simplest, it's a send a command and wait until you get an 'ok' or 'error:' response to send the next one. No delays required. Because of they way it's designed, you should never have any dropped characters. The only drawback is throughput due to USB-serial latency. In my experience, latency isn't a problem with most CNC machines, but can be with lasers in particular. You can also try over-stuffing the serial RX buffer by a character counting scheme, which will help tremendously. 

I also have a streaming script written in Python in the /doc folder of the source code as a template for both methods. It doesn't include how to handle different messages, like status reports or other feedback, but they are all unique and separated by line feed/carriage returns. Each message line should be treated separately. But, it'll get you started.

openhardwarecoza, 2016-06-02:
No problem on the node side,  i'll take care of that.  Here its more a
question of 'just advice on the grbl side'  and 'check that i dont sound
crazy)

chamnit, 2016-06-02:
@openhardwarecoza : As far as Node streamers, there is [GrblWeb's](https://github.com/andrewhodel/grblweb/blob/master/server.js) that you can template from. It's pretty simple overall. In my bench test, this Node streamer works well, but isn't that fast, mainly due to using the simple send-response protocol. Like I said before, USB-serial latency is the main culprit. If modified to use the character-counting scheme, it'll work well for laser purposes.

openhardwarecoza, 2016-06-02:
It was in LaserWeb1 using grblweb's streamer that we had the lost chars
issue..

chamnit, 2016-06-02:
@openhardwarecoza : That's strange. Can you describe what kind of lost characters you experienced and under what circumstances? I do know that Inventables' Easel uses NodeJS for their streamer. They haven't reported any issues to me about the serial stream becoming unstable. 

I do know that Go lang has a garbage collection routine that would kill a serial stream while it runs and return to normal operation. It's uncontrolled and occurs frequently. In my tests with Chilipepper and SPJS, you can hear audible hiccups in the machine when the GC kicks in. Not sure if Node has something similar issue.

openhardwarecoza, 2016-06-02:
I cant be sure,  laserweb started off as a fork of grblweb and reprapweb
put together.  I then started adding all the cam and ui on yop which makes
laserweb as awesome as it is (its even the software of choice on the
kickstarter of the day (fabcreator.com) -  checkout the issue linked above
(we had you in conversation there too)

I couldnt get it fixed,  me and Andrew parted ways,  and i was stuck
without connectivity to Marlin and Grbl.  Or shall i say without reliable
connectivity...   ,so in April i forked myself and created LaserWeb2,
using Serial Port json server.  That was a mistake. I got grbl,  but lost
lasersaur,  smoothie,  repetier and marlin.  Not worth the switch.
Therefore,  now back to node in LaserWeb3

openhardwarecoza, 2016-06-02:
Heres a link to that original issue with grblweb'/reprapweb's code https://github.com/openhardwarecoza/LaserWeb/issues/94

chamnit, 2016-06-02:
Ok. "Read" that old thread. FWIW, I still don't think this is a Grbl problem in particular. In the past, problems like this were due to implementation issues or a problem with the USB-serial chip itself. The reason I say this is because of the feedback I get from both Carbide3d and Inventables. They have thousands of machines in daily operation and haven't heard anything about data corruption issues.

In terms of implementation problems, it's most common that the data is being too fast and the GUI is using a timing-based method like flow-control. Grbl's streaming protocol explicitly does not use timing to ensure data is always sent/received and never lost. Because when you receive a response message, you are guaranteed that last g-code line has been processed and cleared of the serial RX buffer. There should be no chance of data corruption, other than from external influences. 

For the USB-serial problem, it's known that the Chinese CH380 USB-serial converter (common on Chinese Arduino knock-offs) can not be streamed to at a high-rate, otherwise its internal buffers overflow and data gets chopped off. Very similar to what you're seeing. Changing baud rates seemed to help the problem, but never really fix it. I also recall that certain Atmega16U2 USB-serial chips had a buggy set of firmware that could be fixed by updating it.

openhardwarecoza, 2016-06-02:
Definately was not grbl,  nor the hardware,  nor the gcode... Same users
went up and running after switching from nodejs+grblweb code,    to the
spjs backend.  But i cant keep spjs just for grbl,  so need to play until
grbl works with node.    Thats why this time i am taking a clean start.
None of andrews or anyone elses code.  I am asking each of the firmware
vendors,  (you,  marlin,  Smoothieware)  how best to talk tk theirs.  (so
is char count or ok, send, ok best?) so i can write a module,  for eaxh
firmware,  from scratch,  while understanding the needs of that firmware.
No  one module can talk to anything -  that backfired.  I think thats what
destabilised the first nodejs host in laserweb1

openhardwarecoza, 2016-06-02:
Check LaserWeb3 -  seperate threads for each type. (;  -  doing my planning properly

chamnit, 2016-06-02:
@openhardwarecoza : Ok. Well, I can't tell you how to write a streamer, since I've never had to write one. Grbl's v0.9 streaming protocol has some warts (inconsistencies), which requires a GUI writer to account for them. FWIW, I'm trying to fix them for the v1.0 release. I will also write a v1.0 API document to answer most of these questions formally in one place when its released.

As for tips:
- Treat each line from Grbl as a separate message and handle them appropriately based on a handful of types. Generally, it's just an 'ok/error:', '<status message>', and '[ feedback message ]'. The other message types (warts) are usually the welcome string, which GUIs use to set the version, and settings and help messages. Everything that is sent back from Grbl exists in the report.c file.
- You can stream with either the simple streaming or character-counting methods. See interface wiki page for more info.
- Real-time command characters (?,~,!,ctrl-x) DO NOT enter Grbl's serial RX buffer. They are intercepted directly from the serial interrupt register. They don't require any carriage return or line feed. They can be sent at any time, regardless of how full the serial RX buffer is.

Hope that helps.

openhardwarecoza, 2016-06-02:
>    'ok/error:', '', and '[ feedback message ]'. The other message types

That helps a tonne!   I'll do ping pong buffer first,  by far quickest and
easiest,  then investigate character count if and when we need it.

The other tips are great,  i'll check report.c  (in the past i compiled my
regex from the serial log.  Obviously missed a few cases)

Glad i dont need to account for a seperate buffer for realtime commands!

openhardwarecoza, 2016-07-26:
Well,  the Grbl protocol has now been live in LaserWeb3 for about a week and I have three trusted users reporting much success and stability.   Closing this issue!  Thanks for the help!

closed 2016-07-26 by openhardwarecoza