LaserWeb/deprecated-LaserWeb3 #5: Not sure what is wrong

sonex128, 2016-06-05:
Previously had LaserWeb3 installed.  deleted everyting and tried to install new version.  the following is what I get.

C:\>git clone https://github.com/openhardwarecoza/LaserWeb3.git
Cloning into 'LaserWeb3'...
remote: Counting objects: 3009, done.
remote: Compressing objects: 100% (2056/2056), done.
rRemote: Total 3009 (delta 797), reused 2928 (delta 716), pack-reused 0
Receiving objects: 100% (3009/3009), 7.30 MiB | 43.00 KiB/s, done.

Resolving deltas: 100% (797/797), done.
Checking connectivity... done.
Checking out files: 100% (2352/2352), done.

C:\>cd LaserWeb3

C:\LaserWeb3>npm install

> bufferutil@1.2.1 install C:\LaserWeb3\node_modules\bufferutil
> node-gyp rebuild

C:\LaserWeb3\node_modules\bufferutil>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "" rebuild )
gyp ERR! configure error
gyp ERR! stack Error: Can't find Python executable "python", you can set the PYTHON env variable.
gyp ERR! stack     at failNoPython (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\configure.js:401:14)
gyp ERR! stack     at C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\configure.js:356:11
gyp ERR! stack     at FSReqWrap.oncomplete (fs.js:82:15)
gyp ERR! System Windows_NT 10.0.10586
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\LaserWeb3\node_modules\bufferutil
gyp ERR! node -v v4.4.5
gyp ERR! node-gyp -v v3.3.1
gyp ERR! not ok

npm ERR! Windows_NT 10.0.10586
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "install"
npm ERR! node v4.4.5
npm ERR! npm  v2.15.5
npm ERR! code ELIFECYCLE
npm ERR! bufferutil@1.2.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the bufferutil@1.2.1 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the bufferutil package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs bufferutil
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!
npm ERR!     npm owner ls bufferutil
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\LaserWeb3\npm-debug.log

C:\LaserWeb3>

cojarbi, 2016-06-05:
Follow exactly, had the same issue

Clone
Delete Laserweb3 node_modules folder
npm install

sonex128, 2016-06-05:
Thanks.  Working now.

closed 2016-06-05 by sonex128