LaserWeb/deprecated-LaserWeb3 #11: Feed rate

funinthefalls, 2016-06-10:
If you set move rate to 30 in the config, when generating raster gcode it sets the rate to 20mm/m I thought we had changed to mm/s?

funinthefalls, 2016-06-10:
Here is the start of the gcode:
;gcode generated by LaserWeb
; Laser Min: 0%
; Laser Max: 100%
; Black Speed: 10mm/min
; White Speed: 40mm/min
; Resolution (mm per pixel): 0.3527777777777552mm
; Laser Spot Size: 0.2mm
; Laser Feedrate: 30mm/min
; X Offset: 0mm
; Y Offset: 0mm

The feed rates are all mm/m

daned33, 2016-06-10:
Can confirm. Will give a shot fixing.

daned33, 2016-06-10:
Got it working. Made a pull request into master with fix.

funinthefalls, 2016-06-10:
Works great, thanks for the fix.

openhardwarecoza, 2016-06-11:
Gents!  Now you are breaking us!   We multiply what you want in mm/sec on the UI wih ×60 on the gcode si it gets back to mm/min which the firmware needs!   If you don't fix it back to mm/min which is the unit the firmware understands you make our macines 60× slower...

daned33, 2016-06-11:
Uhhh, that's what I did in commit #7668d5a ... Check line 110 in raster.js, I times the value by 60...

The commit fixed the issue.

openhardwarecoza, 2016-06-11:
Alright,  maybe i shouldn't do code review at 4:30am lol.

funinthefalls, 2016-06-11:
I only merged it because it made sense to me...wasn't 4:30 in the morning for me....lol

commit 15421829 (2016-06-10): Merge pull request #12 from daned33/master

Fixed #11. Raster feed rate
commit 7668d5a7 (2016-06-10): Fixed #11 (feedrate for rastergcode)
closed 2016-06-11 by openhardwarecoza