LaserWeb/deprecated-LaserWeb3 #14: SVG Parsing seems to miss every last (or first?) segment of a continuous loop.

thantik, 2016-06-11:
Here's an example of what I'm seeing.  Looks like every last/first segment of my SVG is being dropped.  Off-by-one error?

[image]

openhardwarecoza, 2016-06-11:
Shapes.autoclose...  If we always close then single lines or open ended
lines also gets closed.    But for closed lines and objects you want it to
autoclose.  I. E.  No win win situation of changing the default.  Its
something we have to let users select per object.  Or per file...

I.e something i need to do as part of that tabbed layers cam rework i was
so hyper to start on before the big comms disaster.

Long story short,  pushing ahead with Comms.  Then back to this.

cojarbi, 2016-06-11:
Is there anyway to cheat in the meantime? I need to cut squares but its open.

openhardwarecoza, 2016-06-11:
Use dxf.. Dxf is infinitely more suited format for lasers than svg

openhardwarecoza, 2016-06-11:
[image]

Added to settings -> tools the option to select yes or no...

closed 2016-06-11 by openhardwarecoza

funinthefalls, 2016-06-11:
Good addition

cojarbi, 2016-06-11:
I cant get any code from SVG with the new commit. Can anyone else confirm? Tried enabling and disabling Auto close

thantik, 2016-06-12:
No gcode here either @cojarbi - Additionally, the new option seems to do nothing as far as closing loops.

[image]

cojarbi, 2016-06-12:
Same here. Refreshing after changed setting also does nothing

openhardwarecoza, 2016-06-12:
thats because I am moving the rapid speeds from settings, remember (;

ctrl+shft+i

openhardwarecoza, 2016-06-12:
fixed

thantik, 2016-06-12:
100% confirmed, tested and working!

openhardwarecoza, 2016-06-12:
Great! Last night I twice made capitalization mistakes