LaserWeb/deprecated-LaserWeb3 #16: WIP SD Upload

openhardwarecoza, 2016-06-11:
I added a modal and an upload button... so far.

The upload code is available already from the smoothie project yay: https://github.com/Smoothieware/Smoothieware/blob/8cbd981e85c918e059a6e68d70fbf3cdad0f8ca5/src/libs/Network/uip/webserver/httpd-fs-src/functions.js#L77

openhardwarecoza, 2016-07-23:
@wolfmanjm @arthurwolf

So, i have my Smoothieboard up on network, over the built in web interface I can jog, etc. But uploading any gcode, fails with a 503 response to the XHR request. 

[image]

Browsing http://smoothieboardip/upload:

[image]

Am I missing something?

arthurwolf, 2016-07-23:
Is the firmware recent ?

openhardwarecoza, 2016-07-23:
Build version: edge-c864529, Build date: May 20 2016 04:05:32, MCU: LPC1769, System Clock: 120MHz

arthurwolf, 2016-07-23:
try using another browsers. using chrome right ?

openhardwarecoza, 2016-07-23:
Ah found the problem.  SD card was formatted FAT16, and had too many filed on it. 

When I tried copying a new firmware.bin to it Windows complained about it!   FAT16 only allows 512 files on the root directory (https://support.microsoft.com/en-us/kb/120138) 

[image]

arthurwolf, 2016-07-23:
Yay :)

DarklyLabs, 2016-10-17:
Can the latest ESP-WIFI implementation take advantage of this UI to upload directly to the Smoothie SD?

openhardwarecoza, 2016-10-17:
Negative,  on the next board revision you'll need to add an SD to the ESP

openhardwarecoza, 2016-10-23:
Moved to wishlist on the readme https://github.com/openhardwarecoza/LaserWeb3/commit/b39c56a24a7de76afdbf95dd3e28ae12f6bf6c8c so I can close the issue for now

closed 2016-10-23 by openhardwarecoza