LaserWeb/deprecated-LaserWeb3 #17: Planned Feature in LW3:  Tabbed layers

openhardwarecoza, 2016-06-11:
Layers in threejs for each of the tabs at top of 3d viewer:

gcodeview (hides all objects except the grid, axesgrp, and "object"

All layers - show everything

And then add tabs for each file opened... 

Affect gcode export, (reorder tabs, add remove tabs, etc - should change final gcode)

openhardwarecoza, 2016-06-13:
NB:  As of 2 minutes ago, I deliberately BROKE the CAM on LaserWeb3
I have begun the top down rewrite of the CAM to layers on tabs in the viewer, etc - that nice long term project we wanted for a while now.  
It cant be done without disrupting anything though, massive changes is on order

So what I have done so far:
As you open multiple DXF/SVG files, they all get placed into the viewer and tabs at the top populated with their names.
Closing a tab, removes the object too.
Thats it, all I got to so far...

Next:
Fix the Move/translate/cam so it works "per layer" 

Next:  
Fix the gcode generator so it iterates through all the open tabs and concatenes the gcode into a single file

Next: Complete redesign of how Raster's images are loaded.  Since we need to handle multiple images, and multiple positions for those images - it needs a top down rethink of how the image data is stored before being converted to raster. Current implementation loads new image into old image's place.  New implementation will need to load multiple images and then also allow tabs + cam settings per tab etc﻿

daned33, 2016-06-13:
Couldn't you branch off and leave the master be?

openhardwarecoza, 2016-06-13:
Normally yes,  but some of the testers here have enough trouble with git
pull on a single branch (;

daned33, 2016-06-13:
Kinda defeats the whole point of version control.. 

www.learngit.com

Please lets go with branches. Means that the working cam version can stay unmolested and you can work on the new version without effecting the master. anyone who needs to make pull requests to the dev branch can easily do it in the web interface.

openhardwarecoza, 2016-06-13:
Lol i know git.   (;

@funinthefalls @cojarbi are you guys ready for testing on non-master
branches?

cojarbi, 2016-06-13:
Go ahead.

daned33, 2016-06-13:
Alrighty. 

Master is back on track with a working version of the CAM. You guys can develop on the new branch called 'cam-rewrite'.

Please avoid doing pull requests into the Master before you're sure that the CAM rewrite works. 
Thanks.

funinthefalls, 2016-06-13:
Yup, I an start testing non-master branches. BTW, all the parts came in fore the test rig EXCEPT the X and Y plates Mark said he was sending. And Alex is having issues getting the Y plates to work. Grrrrrrrrrrrrrrrrr.....I want to start building this beast!!!!

openhardwarecoza, 2016-06-13:
Order ex stock from clockwerkmech

openhardwarecoza, 2016-06-22:
Mostly done with the frontend on this.  Just need to rewrite the google drive -> Open file function (right now its duplicated code - maintenance not idea.  Want to rewrite it so it also calles readFile() instead.

openhardwarecoza, 2016-07-07:
Updated the gcode output a little more to accomodate the tabs  and also cleaned up the CAM UI

[image]

[image]

openhardwarecoza, 2016-07-21:
Well, i think we are happy with it by now.  Closing the issue (:

closed 2016-07-21 by openhardwarecoza