LaserWeb/deprecated-LaserWeb3 #18: Can't install Serialport, is there an alternative?

jeffersonbenson, 2016-06-11:
So I have been searching for the past few days and it appears that I am not able to install any version of serialport on my ARMv7 board. I'm trying look around for another way still, but so far it does not appear to be possible. Any ideas of a package that will fulfill the function of serialport, but function in a Debian Jessie environment?

openhardwarecoza, 2016-06-12:
I have it compiled on a. Pine64 and a Pi3.  What board are you on.  Go read
the node serialport readme and make sure you have a working
python/make/gcc/node-pregyp setup

openhardwarecoza, 2016-06-12:
https://github.com/EmergingTechnologyAdvisors/node-serialport#installation-special-cases

closed 2016-06-12 by openhardwarecoza

jeffersonbenson, 2016-06-12:
So I'm trying to compile on a CHIP which uses an Allwinner R8 chip that's an ARMv7 chip. This is the error that I get:
```
2338 verbose rebuildBundles   'object.assign' ]
2339 verbose rebuild bundle /home/chip/Desktop/LaserWeb3/node_modules/serialport/node_modules/node-pre-gyp
2340 info build /home/chip/Desktop/LaserWeb3/node_modules/serialport/node_modules/node-pre-gyp
2341 info preinstall node-pre-gyp@0.6.28
2342 verbose linkStuff [ false,
2342 verbose linkStuff   false,
2342 verbose linkStuff   false,
2342 verbose linkStuff   '/home/chip/Desktop/LaserWeb3/node_modules/serialport/node_modules' ]
2343 info linkStuff node-pre-gyp@0.6.28
2344 verbose linkBins node-pre-gyp@0.6.28
2345 verbose link bins [ { 'node-pre-gyp': './bin/node-pre-gyp' },
2345 verbose link bins   '/home/chip/Desktop/LaserWeb3/node_modules/serialport/node_modules/.bin',
2345 verbose link bins   false ]
2346 verbose linkMans node-pre-gyp@0.6.28
2347 verbose rebuildBundles node-pre-gyp@0.6.28
2348 verbose rebuildBundles [ 'mkdirp',
2348 verbose rebuildBundles   'nopt',
2348 verbose rebuildBundles   'npmlog',
2348 verbose rebuildBundles   'rc',
2348 verbose rebuildBundles   'request',
2348 verbose rebuildBundles   'rimraf',
2348 verbose rebuildBundles   'semver',
2348 verbose rebuildBundles   'tar',
2348 verbose rebuildBundles   'tar-pack' ]
2349 info install node-pre-gyp@0.6.28
2350 info postinstall node-pre-gyp@0.6.28
2351 info install serialport@3.1.2
2352 verbose unsafe-perm in lifecycle false
2353 info serialport@3.1.2 Failed to exec install script
2354 verbose node_modules/serialport unbuild
2355 warn This failure might be due to the use of legacy binary "node"
2356 warn For further explanations, please read
/usr/share/doc/nodejs/README.Debian
2357 info preuninstall serialport@3.1.2
2358 info uninstall serialport@3.1.2
2359 verbose true,/home/chip/Desktop/LaserWeb3/node_modules,/home/chip/Desktop/LaserWeb3/node_modules unbuild serialport@3.1.2
2360 verbose /home/chip/Desktop/LaserWeb3/node_modules/.bin,[object Object] binRoot
2361 info postuninstall serialport@3.1.2
2362 error serialport@3.1.2 install: `node-pre-gyp install --fallback-to-build`
2362 error Exit status 1
2363 error Failed at the serialport@3.1.2 install script.
2363 error This is most likely a problem with the serialport package,
2363 error not with npm itself.
2363 error Tell the author that this fails on your system:
2363 error     node-pre-gyp install --fallback-to-build
2363 error You can get their info via:
2363 error     npm owner ls serialport
2363 error There is likely additional logging output above.
2364 error System Linux 4.4.11-ntc
2365 error command "/usr/bin/nodejs" "/usr/bin/npm" "install" "serialport" "--build-from-source"
2366 error cwd /home/chip/Desktop/LaserWeb3
2367 error node -v v0.10.29
2368 error npm -v 1.4.21
2369 error code ELIFECYCLE
2370 verbose exit [ 1, true ]
```
Just a snippet of npm-debug.log since I'm using screen instead of ssh. Recommend reopening this issue.

openhardwarecoza, 2016-06-12:
Please open the issue on node-serialport,  this isnt a LaserWeb3 issue (:

jeffersonbenson, 2016-06-12:
Will do!