LaserWeb/deprecated-LaserWeb3 #19: Incomplete Config if you leave laser on and laser off commands blank in the settings

funinthefalls, 2016-06-12:
If you leave the on and off command settings blank, it gives an !incomplete config error.

funinthefalls, 2016-06-12:
This is the contents of my settings backup json file:
{"autoClose":"Yes","debug":"undefined","endgcode":"G28","homingseq":"G28","imagePosition":"TopLeft","laserXMax":"330","laserYMax":"230","lasermultiply":"255","laseroff":"","laseron":"","lastUsedBaud":"250000","lastUsedPort":"COM3","loglevel":"ERROR","rapidSpeed":"undefined","rapidspeed":"40","smoothieIp":"","spotSize":"0.1","startgcode":"G28","subnet1":"","subnet2":"","subnet3":"","tour_current_step":"0","tour_end":"yes","useNumPad":"Disable","useOffset":"Disable","useVideo":"Disable"}

There appears to be two entries for rapidspeed, first one says "undefined", second one says "40"

openhardwarecoza, 2016-06-12:
Fixed

closed 2016-06-12 by openhardwarecoza