LaserWeb/deprecated-LaserWeb3 #24: Constructor error when attempting to open port

jeffersonbenson, 2016-06-16:
This is mostly to get the ball rolling as I do some debugging on my end. After finally getting the software to compile in npm with no issues, I opened the web frontend to tried to open all the different ports to get something to connect to the GRBL pre installed (v0.9) on my Azteeg Gradus M1 Pro. However, I'm greeted with the following error:
```
Missing error handler on `socket`.
TypeError: serialport is not a constructor
    at Socket.<anonymous> (/home/chip/Desktop/LaserWeb3/server-grbl.js:97:14)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at Socket.onevent (/home/chip/Desktop/LaserWeb3/node_modules/socket.io/lib/socket.js:335:8)
    at Socket.onpacket (/home/chip/Desktop/LaserWeb3/node_modules/socket.io/lib/socket.js:295:12)
    at Client.ondecoded (/home/chip/Desktop/LaserWeb3/node_modules/socket.io/lib/client.js:193:14)
    at Decoder.Emitter.emit (/home/chip/Desktop/LaserWeb3/node_modules/socket.io/node_modules/socket.io-parser/node_modules/component-emitter/index.js:134:20)
    at Decoder.add (/home/chip/Desktop/LaserWeb3/node_modules/socket.io/node_modules/socket.io-parser/index.js:247:12)
    at Client.ondata (/home/chip/Desktop/LaserWeb3/node_modules/socket.io/lib/client.js:175:18)
    at emitOne (events.js:96:13)
```
Any thoughts? Everything's up to date as far as I can tell with "npm list"

openhardwarecoza, 2016-06-16:
Issues are for bugs and development...  Stay up to date on news on the
google community please
https://plus.google.com/102920303032765220786/posts/SPvskweUAGd  -  read my
comments below.   LaserWeb3 is 10 days from its birth. No where near done
and grbl is low priority.    Also  did you check the other open issues here
before you clicked New Issue?  At all?  If you are a developer,  welcome,
fork and help get it going.  If not,  go over to
http://openhardwarecoza.github.io/LaserWeb2

jeffersonbenson, 2016-06-16:
Thank you Peter. Sorry if this isn't the place to be asking such questions. After doing much searching across multiple projects I deliberated enough to open up an issue. I myself am a developer and tinkerer. I shall nevertheless pursue forward and try to make the best use of the system. Thank you for the help. Again humblest apologies. 
Jeff

closed 2016-06-16 by jeffersonbenson

openhardwarecoza, 2016-06-16:
No problem (: -  just please actively participate in the community,  then
you wont need to wonder about whats going on.  The Google Plus community
has daily updates.

openhardwarecoza, 2016-06-16:
I went and updated everything in huge capitols to avoi confusion,  and put
pinned posts on the community...