LaserWeb/deprecated-LaserWeb3 #27: HTTPS Support

openhardwarecoza, 2016-06-23:
Mac requires HTTPS for getUserMedia() - so camera doesnt work on http...

emteeoh, 2016-07-22:
Google Chrome won't allow the camera to work over http, *EXCEPT* when connecting to 127.0.0.1 So this is not a Mac-only issue.

emteeoh, 2016-07-27:
Um... I think you might have misunderstood me, Peter. In my setup, I have LW3 running on a Pi, but access it via chrome on my laptop. AFAIK, if LW3 was running on my laptop, I could use webRTC by going to http://localhost:8000/, or http://127.0.0.1:8000/, or http://192.168.2.33:8000/. They're all fine. What I *cannot* do is use the laptop webcam with LW3 on the pi, via unencrypted http. Personally, I'd like to be able to connect the camera to the pi, not the laptop, but I have not figured out how to do that using webRTC yet. Admittedly, I have not put much effort into it yet.

openhardwarecoza, 2016-07-27:
My title change was actually from researching the RFC.   By standard,  only
http://localhost and http://properdomain.tld  should work.  Anything else
is a fluke,  or need https.

openhardwarecoza, 2016-07-27:
Re remote camera,    we did it in LaserWeb1 using mjpeg streamer,  but its
not as nice as webrtc.    The thing to me is,  why pi?  Its slow,  you
anyway have a pc to cad on,  why suffer.

emteeoh, 2016-07-27:
Re: title change. Oh! nevermind then!

re: camera. I don't want to roll back the clock. I'm hoping to figure out how to hook something "server side" into webRTC. 

Re: pi... I'm looking to do one single shared "server" to manage the laser cutter. I don't want to have to move USB connections around when different people are using different computers for different purposes. Also, the CAD station near the cutter is a very under-powered machine itself, and is really only used to prep the drawings. Its also not the most reliable computer ever. Basically, I'm trying to not rely on it.
Just to be clear, its a pi3, not an original pi, so its not too slow, and so far in testing, it works pretty well. I don't feel like pages are reloading slow or anything. I'm not sure I'm suffering. The lack of "server-ish" features to LW3 is the closest I come, and its manageable. So far, I'm only really supporting 2 people across 3 computers. That number may grow, but not by much. Its not a privately owned workshop, so I really only worry about me, the owner, and his wife.

openhardwarecoza, 2016-08-01:
Added a Remote Webcam feature that can grab frames off an HTML link instead

closed 2016-08-01 by openhardwarecoza