LaserWeb/deprecated-LaserWeb3 #33: Display path size

cojarbi, 2016-07-04:
Show the path size. Want to make sure im scaling to the correct size

openhardwarecoza, 2016-07-07:
Elaborate?  Already displaying that on the Quotes tab

cojarbi, 2016-07-07:
I was referring to the path or raster box / perimeter size. Yes we should have the scale allready but showing the info is useful specially if I scale in LW

openhardwarecoza, 2016-07-10:
You mean like this?  

[image]

(Works for any object, raster or vector):  When you click that objects's tab:  We can display more parameters around the dashed line bounding box?

cojarbi, 2016-07-10:
Thanks,  looks like the perfect place. 

AY

closed 2016-07-10 by openhardwarecoza