LaserWeb/deprecated-LaserWeb3 #36: Gcode created out of sequence

funinthefalls, 2016-07-04:
Here is the first part of the gcode file:
;Gcode generated by Laserweb Three2Gcode;
G28
G28
;End of file
; Raster:
; Laser Min: 10%
; Laser Max: 60%
; Black Speed: 4200mm/s
; White Speed: 1200mm/s
; Resolution (mm per pixel): 0.3527777777777552mm
; Laser Spot Size: 0.1mm
; Laser Feedrate: {6}mm/s
; X Offset: 0mm
; Y Offset: 0mm 

G1 FNaN
G0 Y172.861
G0 X0.000 S0
G1 X0.706 S132 F3717
G1 X1.058 S131 F3672
G1 X1.411 S129 F3636
G1 X1.764 S131 F3672

openhardwarecoza, 2016-07-07:
This happens becuase the Rasters are generated out of order as seperate threads - they get appended whenever its finished

[image]

Busy changing the backend like this:

1.  It adds the start gcode
2.  If a GCODE file was opened it adds that next
3.  Then it adds the DXF/SVG paths
4.  Add end gcode
5.  Then as the background raster jobs finish, inject them after the paths, but before end gcode  (I know @cojarbi wants this the other way round - will try and get it like that - ie raster first, cut after)

funinthefalls, 2016-07-08:
Will test today.

closed 2016-07-10 by openhardwarecoza