LaserWeb/deprecated-LaserWeb3 #37: First G1 code has no G1 and F parameter

ahorn42, 2016-07-08:
The first 3 lines of my generated GCode looks like this:

	G0 F3000 X15.750 Y0.345
	 X15.750 Y0.345 Z0
	G1 F1200 X15.750 Y0.345 Z0 S1.00

In the second line it's missing `G1 F1200` at the beginning. Note also the line 2 starts with a space character.

funinthefalls, 2016-07-08:
Can you please attach the file you were working with?

ahorn42, 2016-07-09:
Till now it happened with all file, but here is one to test [ahorn.svg](https://owncloud.philippberger.de/index.php/s/JaTQSXUyun6SV6s) and the generated gcode when scaled at 500dpi [ahorn.gcode](https://owncloud.philippberger.de/index.php/s/vISE0c3wqes8zX5)

openhardwarecoza, 2016-07-09:
Settings -  gcode -  travel speed.  I suspect you left it blank?

ahorn42, 2016-07-09:
Uhm no, it's set to 50 mm/sec - and it does not look like an placeholder text. Furthermore I discovered that every first G1 after a G0 misses its G1 and F parameter.

openhardwarecoza, 2016-07-09:
Heading home now after a 72hour workweek.  Will check tomorrow

openhardwarecoza, 2016-07-10:
Checked, and code updated. Please 'git pull' test and let me know if its resolved (:

closed 2016-07-10 by openhardwarecoza

ahorn42, 2016-07-10:
Yep, now it works :)

Thanks a lot for this fast fix!

openhardwarecoza, 2016-07-10:
Keep the testing and cool ideas coming  (; i'll keep fixing and adding