LaserWeb/deprecated-LaserWeb3 #40: Export not exporting revised gcode

r0773nluck, 2016-07-12:
Testing out editing gcode before saving when i delete lines in gcode and then hit save it still puts the lines back in the saved gcode. it also didnt do this everytime it did work once when i deleted the lines and saved. 

side note can a check box be added that adds a G92 command and deletes the first 3 G0 commands so if available
[image]
 a person can set laser position manually with laser guidance from a memory card?

openhardwarecoza, 2016-07-12:
The gcode tab is a preview and export.  Not a Edit.    To edit gcode,
revise the cam settings,  or your start/end gcode.

In laserweb: the moves are stored in threejs and converted to gcode for
display,  export  and job execution.

Not from the text fields,  but from the threejs userData attributes.

I' change the textareas to 'disabled'  to avoid the confusion,  else people
will think they cab edit it

openhardwarecoza, 2016-07-12:
Changed UI to prevent 'editing attempt'

[image]

openhardwarecoza, 2016-07-12:
Oh forgot to reply to:  "side note can a check box be added that adds a G92 command and deletes the first 3 G0 commands so if available"

-  How I use G92:
My start gcode contains nothing (my machine doesnt have endstops wired yet, so no g28)
Then i manually jog to lower left corner and press the Set Zero button (next to play/stop, under the DRO) - that G92's out all three axes

Then I press play - all moves are then from current machine position =   0,0 in viewer

closed 2016-07-12 by openhardwarecoza

openhardwarecoza, 2016-07-12:
Saw a missing number in G0 on your screenshot (NaN mean Not a Number) -  fixed it in https://github.com/openhardwarecoza/LaserWeb3/commit/3895e4786fc57ba61306cf63b731f3ceed193a3c - git pull to update please

r0773nluck, 2016-07-12:
Ok I'll git pull again but even if I g92 before each print I need to make sure to delete the first line G0 Y46 other wise it's will always move to the offset of your print unless I am missing a setting somewhere that will get rid of that

openhardwarecoza, 2016-07-12:
The first G0 Y46 moves the head to the top of the IMAGE, before it starts moving left to right, since we raster from top, to bottom on rasters...

openhardwarecoza, 2016-07-12:
so in your example you have a 47mm high image,   from 0, it moves to y 47, then works its way back to 0,0 during the raster

r0773nluck, 2016-07-12:
Ok that makes sense.
 Maybe in a future update (I really don't know the difficulty of it) could this be an option to disable that move as it seems, at least to me, more intuitive to start the head in the position where you want it to start engraving from the top that way if your engraving an object bigger then the image you don't have to guess if it's lined up low enough on that object to not miss an engrave. If that makes any sense

openhardwarecoza, 2016-07-12:
It doesnt make any sense if you add a raster and vector in same job.  If
you do,  your logo will sit offset from where it is in the viewer,  on the
final cut piece.

Start thinking like a machine.  0,0 = lower left.     Thats the Origin of
the Machine!  If you burn an image,  line up lower left.  Thats what makes
sense

openhardwarecoza, 2016-07-12:
Your comment literally made me facepalm :(

r0773nluck, 2016-07-12:
I get the think 0,0. This option would not be for combination of raster/cut. This would be for if a person wants to engrave just some random item. I position the item on the print bed line my laserhead with my cross hairs where I want it to begin and hit print. Also keep in mind I'm on a K40 so for some dam reason they put the 0,0 in the top left instead of bottom left

openhardwarecoza, 2016-07-12:
No,  after conversion,   0, 0 should be lower left.   Is the existing Y
endstop as a MAX endstop.   Think we just found your problem. You are
working in reverse!!!

Fix it they way the Cartesian number system Should Be!  0,0 lower left.

r0773nluck, 2016-07-12:
All my setting are correct I did no hardware conversion to move end stops so my x and y end stops are still in the original position so when the unit homes it moves to those enstops. I spoke with Ariel Yahni who sent me hi is config which stoped the machine from crashing and printing Images in reverse. So everything with your program is working with my machine including the circle fix. Thank you for the fix to the extra g0 command and making the windows uneditable to avoid confusion for new users.

I guess what I'm saying isn't conveying in writing so never mind I'll just continue to manually clear out the G0 Y command

openhardwarecoza, 2016-07-12:
You'll run into issues with that eventually.    Sleep on it.  I promise
you,  you need a mind shift.

It just starts rastering from the top.  But the 'image'  and 'machine'
share lower left as origin.  End of story