LaserWeb/deprecated-LaserWeb3 #41: "End G-Code" not updateable once set

pda3k, 2016-07-17:
Once the setting "End G-Code" is set, it can't be updated.
[image]

openhardwarecoza, 2016-07-17:
Change it in Settings -  gcode.

Partial screenshot doesn't show me if you are on settings or on thee gcode
preview tab.  Gcode preview is read only,  so i assume so.

pda3k, 2016-07-17:
That's what I did.  I went in to update the End Gcode settings using the gears icon on the left side of the screen.  Other settings to include the "Start Gcode" were updateable but not the End Gcode. Is there another way to do this?

openhardwarecoza, 2016-07-17:
I'll check in the morning.  If you want to dig into the raw data,  open
Developer Tools in chrome (ctrl+shift+i,  go to Resources (tab on dev
tools  or sometimes hidden under the three upright dots menu)  ,
LocalStorage, 127.0.0.1,   and you should find a field called endgcode to
edit / delete

pda3k, 2016-07-17:
That did the trick.  Get some sleep - coding makes you go blind  ;)

openhardwarecoza, 2016-07-18:
I tested on my side, and not able to replicate:

1. I set an end gecode, and saved
2. I refresh page, tried editing, still could
3. Closed browser, came back, could still edit
4. generated gcode, tried editing from settings, still could.

pda3k, 2016-07-18:
> I tested on my side, and not able to replicate:
>
> I set an end gecode, and saved
> I refresh page, tried editing, still could
> Closed browser, came back, could still edit
> generated gcode, tried editing from settings, still could.

Odd that this is the only input that I can't seem to change. I am using the latest version of Chrome on Win 10. I'll check out a few other things on my end and see if I can isolate the problem. Thanks for looking at it. 

Phil

openhardwarecoza, 2016-07-18:
I'm on chrome,  win10 too.  Do a git pull to latest code,  and try clearing
browser cache

openhardwarecoza, 2016-07-21:
Closing for now, neither of us can replicate it so looks to be a PC bug.

pda3k, 2016-07-22:
I've been on a business trip all this week so I could not do any further testing/investigating.  I'll check it out this weekend and let you know if I find anything.  Thanks
Phil

Date: Wed, 20 Jul 2016 22:55:32 -0700

pda3k, 2016-07-28:
Peter,
I think I may have been able to re-create what I experienced a few weeks ago using the Chrome browser in Win 10.  I was able to clear up the problem by clearing all cookies and cache (basically starting over with all settings).  Somehow, when I had the problem, I also had enabled "Use Touch Numpads".  I wondered why I was seeing the keypad keyboard but it didn't dawn on me that the two were related.  Once I turned off the touch numpads, everything worked fine.  When I turned them back on, only a few of the input boxes would accept the changes.  The text input boxes seemed to be the issue.  I tried the Edge browser and IE 11 but got different results (somewhat worked).  I know Chrome works best but I wanted to see if other browsers handle this differently and they do.  I also have a PIPO and I got the same results while using the touchscreen/touchpad to try to change data.
Let me know if you see the same thing on your end.  
BTW - I am putting a few bucks in your tin cup to help out.  Thanks for all the great work you do on this project.
Phil
Date: Wed, 20 Jul 2016 22:55:32 -0700

openhardwarecoza, 2016-07-28:
Aha,  i recall the touch numpads had a isNaN (not a number error)  that popped up sometimes.  Might relate

openhardwarecoza, 2016-07-29:
OK, I updated a few bits in the jquery numpad plugin, do a git pull, test and see if the issue still occurs?

pda3k, 2016-07-29:
Will do...

Phil

pda3k, 2016-07-30:
Almost there.  I am now able to edit and save changes.  What it is not doing is populating the touch pad keyboard with the old settings when you first click on the textbox (it comes up blank and you must re-enter anything that might have already been in the textbox).

openhardwarecoza, 2016-07-30:
Alright,  thats sort of expected behaviour (:

closed 2016-08-01 by openhardwarecoza