LaserWeb/deprecated-LaserWeb3 #43: Please update README.md#supported-firmwares

trillerpfeife, 2016-07-22:
Thanks for your great work!
Yesterday i was a little bit confused because GRBL doesnt seem to be supported but than i saw the GRBL commit! Yeah. Pleas update https://github.com/openhardwarecoza/LaserWeb3#supported-firmwares so that other people can find it right away.

openhardwarecoza, 2016-07-22:
I just want the community to do some testing before i declare it
'supported',  can we give it a day or two?  The commit log usually reads
like a book though

trillerpfeife, 2016-07-22:
ok thx

closed 2016-07-22 by trillerpfeife

openhardwarecoza, 2016-07-23:
Updated.  Have three users reporting success.

trillerpfeife, 2016-08-02:
Mine is also working fine! But pause and abort the job is nor working reliable.

openhardwarecoza, 2016-08-02:
Pause and abort,  stops sending commands.  But whatever already left the
wire,  ie ended up on the serial buffer of the controller. Should still
take a few moves to flush.  Does that Match your observation?