LaserWeb/deprecated-LaserWeb3 #44: Error sending gcode

diegotrap, 2016-07-24:
I am having problems sending this gcode. I use the GRBL server and the code has been generated with the new laser multipass. I had no problems with LaserWeb3 before this one, when I updated from commit 9f2ff68079b663c2c389c08395d77ae43ae2e387 (when GRBL support was added)  to the last commit (version bump, commit 145509bc18b2aca12671e093ad328f389502df8f).

The problem is that you hit the Run button and machine doesn't move. It may happen when using the new multipass. I've noticed that the first line of the gcode is "undefined".

[file.gcode.zip](https://github.com/openhardwarecoza/LaserWeb3/files/380526/file.gcode.zip)

openhardwarecoza, 2016-07-24:
Do you have start gcode at all?  In settings -  GCODE? I think we force add startgcode since #36

diegotrap, 2016-07-24:
I don't have start gcode, but I am being able to make it work if I only set one pass.

I have noticed that the problem only happens when I set more than one pass in the new multipass feature. With one pass the laser works fine, with more than one it generated this (notice the undefined in the first line):

[image]

openhardwarecoza, 2016-07-24:
Aha,  that screenshot shows the issue much better!  Its not in the
startgcode,  but in first line of toolpath. Ok,  let me check

openhardwarecoza, 2016-07-24:
Please test this fix.

diegotrap, 2016-07-24:
Tested, it still generates an "undefined" in the first line of the CAM code.

In the console log the error is reflected as:

> Recv: ok
> Sent: undefined Q: 43
> Recv: error: Bad number format
> Nope

openhardwarecoza, 2016-07-24:
All right,  calling it a night,  will debug in the morning while i sit in
front of a PC.   Good thing we know exactly what (multipass)  and where
(first line of toolpath)

diegotrap, 2016-07-24:
The new commit doesn't generate the undefined code (not in multipass and not in single pass) but it doesn't support multipass. Even if you introduce more than one passes, it only does one. Path length is the same for one or several passes and if you send it to the machine, it only completes the first pass.

cojarbi, 2016-07-25:
Commit be22e3f will connect and work perfectly
Commit c3ab920 connect but -> Sent: undefined Q: 91Recv: error:Unsupported command -undefined )
Commit 01f3a6f to adf3cc2  Error connecting bellow

Statement in terminal:

Missing error handler on `socket`.
TypeError: SerialPort is not a function
    at Socket.<anonymous> (/Users/ayahni/LaserWeb3/server-smoothie.js:159:14)
    at emitOne (events.js:77:13)
    at Socket.emit (events.js:169:7)
    at Socket.onevent (/Users/ayahni/LaserWeb3/node_modules/socket.io/lib/socket.js:335:8)
    at Socket.onpacket (/Users/ayahni/LaserWeb3/node_modules/socket.io/lib/socket.js:295:12)
    at Client.ondecoded (/Users/ayahni/LaserWeb3/node_modules/socket.io/lib/client.js:193:14)
    at Decoder.Emitter.emit (/Users/ayahni/LaserWeb3/node_modules/socket.io/node_modules/socket.io-parser/node_modules/component-emitter/index.js:134:20)
    at Decoder.add (/Users/ayahni/LaserWeb3/node_modules/socket.io/node_modules/socket.io-parser/index.js:247:12)
    at Client.ondata (/Users/ayahni/LaserWeb3/node_modules/socket.io/lib/client.js:175:18)
    at emitOne (events.js:77:13)

openhardwarecoza, 2016-07-25:
> at Socket. (/Users/ayahni/LaserWeb3/server-smoothie.js:159:14)

@cojarbi -  latest commits also removed serialport. Serialport -  update
your node modules

openhardwarecoza, 2016-07-25:
@cojarbi - i updated our code to match upstream
https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/871

openhardwarecoza, 2016-07-25:
Pushed a fix,  i tested myself.

closed 2016-07-25 by openhardwarecoza

cojarbi, 2016-07-25:
All good here on my side after clean npm install

diegotrap, 2016-07-25:
Everything works great. Thanks for your work. Now it's a delight to use your software.

openhardwarecoza, 2016-07-25:
Only now?  Lol.    Close this one when you guys are happy.  Not to self,
dont code on your phone!