LaserWeb/deprecated-LaserWeb3 #45: Tabbing off y-offset value in bitmap panel resets offsets to 0

krolco, 2016-07-25:
I tried this on Safari 9.1.1 and Chrome Version 51.0.2704.103 (64-bit), using version 30170 of LW3.

In the CAM:Convert to G-CODE Panel, when I am editing the settings for a bitmap to be engraved, tabbing off the "Y" box in the "Position Offset" section into the "Bitmap Resolution" box causes both the X and Y values in Position Offset to reset to 0.

openhardwarecoza, 2016-07-25:
I think its because we monitor keyup on the resolution,  and resolution change is supposed to put it back to 0,0 (so we can redraw the mesh with the new resolution).    If you click out,  instead of Tab,  it works correct right?   In that case I would think the best fix is then to swop their positions,  resolution at the top,  offset below it.  Makes more workflow sense too. Set the size,  then place it

krolco, 2016-07-25:
Hi, yes, it works if I click. If you move the resolution to the top, it would also be consistent with the "Trace to Vector" output panel (which has the resolution as the first element also)

krolco, 2016-07-26:
I moved the DPI selection to be the first option on the panel and put the offset right after (since it's likely to be the first things you set) and I moved the trace option to the final option on the panel (since it creates a new tab afterwards). I also moved the offset position to the first option on the DXF panel for consistency; it looks like it was already in that position on the CNC panel. That seemed to fix the problem as you figured above. As soon as I can figure out how to make a pull request, I'll send it in.

openhardwarecoza, 2016-07-26:
Agreed (:  -  will do that today then (:

krolco, 2016-07-26:
I opened pull request #49 for this.

openhardwarecoza, 2016-07-26:
Ahh sorry, I didnt see the PR

krolco, 2016-07-26:
The pull request is for the offset-reset issue. I didn't investigate the Safari issue yet, but if you don't intend to support Safari, that's OK -- I am fine using Chrome, and I'm not particularly good at investigating browser differences. :) If you don't plan on supporting this, you can just close the issue.
Thank you!
Monte

Date: Mon, 25 Jul 2016 21:40:33 -0700

krolco, 2016-07-26:
Thanks, I saw how you merged it manually. I updated my fork and closed the pull request. No problem on the Trace button -- I liked it better at the bottom, but I hadn't considered the interaction with the parameters :) I think this is all resolved now, so I'll close this. Thanks again!

closed 2016-07-26 by krolco