LaserWeb/deprecated-LaserWeb3 #48: Bitmap Preview and Trace Preview not working on Safari 9.1.1

krolco, 2016-07-26:
I don't know if this is known, but the bitmap preview and trace views don't work on Safari 9.1.1 on OS X 10.11.5. I tested with several different bitmaps; they work on Chrome. I've attached one of the affected bitmaps. 
[image]

openhardwarecoza, 2016-07-26:
I dont have access to OSX,  does safari have a Javascript Dev Console?
"(Same as Chrome's Ctrl+Shift+I console)?  If yes can you see if there pops
up any errors?  As a sidenote,  since January we've always said we only
support Chrome.  Since so much of the code relies on the Javascript
engine,  and Chrome's js engine and webgl support is always ahead of the
curve

krolco, 2016-07-26:
I'll check for an JS errors, but if you intend Chrome only, you can just close this.

openhardwarecoza, 2016-07-26:
For now, if you dont mind, lets.  With the whole nodejs thing being easy to cross platform, and chrome being great across all platforms, if we can it'd be great

openhardwarecoza, 2016-07-26:
You can run a quick check against http://kilobtye.github.io/potrace/  to see if its potrace's code,   or LaserWeb - that conks out (:

krolco, 2016-07-26:
Something goes wrong loading the bitmap -- I see "Sanity Check" output from raster.js:69 that shows the wrong size for the bitmap (0,0), even though the Safari debugger shows the right size for it. Everything else after that goes bonkers because it computes a zero-area bounding box. I might try to look a bit deeper.

closed 2016-07-26 by krolco

openhardwarecoza, 2016-07-26:
Aha,  so its not on the trace code,  but rather up near the file open.
Check the readFile function in main.js

krolco, 2016-07-29:
Fyi, I just submitted PR #58 for this.

openhardwarecoza, 2016-07-29:
Merged (: