LaserWeb/deprecated-LaserWeb3 #51: Rotary Support (Work in progress)

openhardwarecoza, 2016-07-27:


openhardwarecoza, 2016-07-27:
@HakanBastedt  (;

openhardwarecoza, 2016-07-27:
Here's an example rotary file (Where Y was converted to A)

```
(G Code Wrapped by the Chestnut Pens GCodeWrapper Program.)
(Toolpath Creation Program © R.W.Williams, 2013)
(Max Straight segment length = 1.0000mm)
(Max arc angle of segment around centre = 10.0000degrees)
(Note: All dimensions are converted to mm)
(Total Running Time minimum = 40 Minutes, 29.19 Seconds. )
(Original file Axis height Z = 4.0000)
(Original file Nominal Work height Z = 7.5000)
(This file Axis height Z = 0.0000)
(Y axis is wrapped...)
(...onto the A axis)
G0 F3600 A-998.2889 X0.0000
G1 F1200 Z-4.0000
M3
G1 F1200 X18.5400 S255
G1 F1200 A-257.9456 S255
G1 F1200 X47.4760 S255
G1 F1200 A-16.0756 S255
G1 F1200 X0.0000 S255
G1 F1200 A-998.2889 S255
M5
G0 A-497.1309 X69.9860
G1 F1200
M3
G1 F1200 A-524.6002 X54.0650 S255
G1 F1200 A-585.6611 X55.1650 S255
G1 F1200 A-632.1362 X56.6640 S255
G1 F1200 A-668.7400 X58.7360 S255
G1 F1200 A-700.1381 X61.5550 S255
G1 F1200 A-718.5709 X64.2770 S255
G1 F1200 A-732.3055 X67.8580 S255
G1 F1200 A-743.6992 X76.5350 S255
G1 F1200 A-740.2615 X83.2880 S255
G1 F1200 A-729.9646 X88.5680 S255
G1 F1200 A-708.7652 X92.7320 S255
G1 F1200 A-672.6688 X96.1390 S255
G1 F1200 A-636.2451 X98.0420 S255
G1 F1200 A-587.2490 X99.4950 S255
G1 F1200 A-533.2273 X100.4160 S255
G1 F1200 A-481.7265 X100.7230 S255
G1 F1200 A-167.5001 S255
G1 F1200 A-88.7757 X101.1120 S255
G1 F1200 A-16.0756 X102.8100 S255
G1 F1200 X87.1760 S255
G1 F1200 A-57.9506 X85.9480 S255
G1 F1200 A-103.1815 X85.3750 S255
G1 F1200 A-58.9655 X82.1110 S255
G1 F1200 A-29.4828 X78.8680 S255
G1 F1200 A-7.3666 X74.0890 S255
G1 F1200 A0.0000 X68.5950 S255
G1 F1200 A-3.6833 X64.9410 S255
G1 F1200 A-14.7332 X61.7550 S255
G1 F1200 A-33.1661 X59.0370 S255
G1 F1200 A-58.9655 X56.7870 S255
G1 F1200 A-90.1508 X55.0230 S255
G1 F1200 A-124.7902 X53.7640 S255
G1 F1200 A-162.8510 X53.0080 S255
G1 F1200 A-204.3495 X52.7560 S255
G1 F1200 A-278.2119 X53.4820 S255
G1 F1200 A-337.6850 X55.6620 S255
G1 F1200 A-362.1257 X57.4220 S255
G1 F1200 A-383.2433 X59.7950 S255
G1 F1200 A-415.3944 X66.3850 S255
G1 F1200 A-458.6118 X78.5400 S255
G1 F1200 A-491.1067 X84.4340 S255
G1 F1200 A-533.8166 X84.1170 S255
G1 F1200 A-561.4495 X83.1650 S255
[... 1748 lines cut ...]
G1 F1200 A-128.5881 X10.9790 S255
G1 F1200 A-129.6685 X11.0820 S255
G1 F1200 A-130.8308 X11.1830 S255
G1 F1200 A-132.0586 X11.2790 S255
G1 F1200 A-133.3682 X11.3720 S255
G1 F1200 A-134.7597 X11.4610 S255
G1 F1200 A-136.2166 X11.5460 S255
G1 F1200 A-137.7391 X11.6260 S255
G1 F1200 A-139.3106 X11.7010 S255
G1 F1200 A-140.9640 X11.7720 S255
G1 F1200 A-142.6501 X11.8380 S255
G1 F1200 A-144.4017 X11.8980 S255
G1 F1200 A-146.2025 X11.9530 S255
G1 F1200 A-148.0359 X12.0030 S255
G1 F1200 A-149.9021 X12.0470 S255
G1 F1200 A-151.8174 X12.0860 S255
G1 F1200 A-153.7491 X12.1190 S255
G1 F1200 A-155.7136 X12.1450 S255
G1 F1200 A-157.6944 X12.1660 S255
G1 F1200 A-159.6915 X12.1810 S255
G1 F1200 A-161.6887 X12.1900 S255
G1 F1200 A-163.7022 X12.1930 S255
M5

```

wolfmanjm, 2016-07-27:
@openhardwarecoza The new step generation is done and merged in. Although the E is now treated as a real axis, I did not have any plans to add an A axis per se, as I do not have a 4 axis machine and I have no way to test such a thing. Someone else may have to figure out how to do that, although making the system handle an A instead of E is pretty easy, and I suppose I could be persuaded to do that at least.
My plans were to do the s curve acceleration next, not n-axis. (although the code base is a lot closer to being able to do n-axis than it used to be).

openhardwarecoza, 2016-07-27:
No rush @wolfmanj, building the viewer and cam will take a lot longer - so keep on working on S-Curve so we can rub it into TinyGs wounds (; 

Maybe by then we can get the community to buy you a rotary axis laser (;

openhardwarecoza, 2016-07-27:
Just tagging so the issues ref across:   Smoothie XYZABC  support: https://github.com/Smoothieware/Smoothieware/issues/120

wolfmanjm, 2016-07-28:
@openhardwarecoza FWIW I have created a Pull request that adds A B C, the A is raw but I think will do what you want, if you set steps_per_mm to steps_per_degree and pass in the raw degrees it should work. I have no way to test it though.
https://github.com/Smoothieware/Smoothieware/pull/989

openhardwarecoza, 2016-07-28:
> Smoothieware/Smoothieware#989

I'll get the test team to start making plans!   @funinthefalls @cojarbi
awesomeness!

cojarbi, 2016-07-28:
Why o why this developments happens when I'm away

AY

openhardwarecoza, 2016-07-28:
Hahaha.  You should go away more often (;

openhardwarecoza, 2016-07-28:
@hakanbastedt

openhardwarecoza, 2016-07-29:
https://cloud.githubusercontent.com/assets/11662298/17242154/e3e9aae8-5575-11e6-9ab9-c48e1ac8d1b9.png

+HakanBastedt added this for us too (:  Maths done, no on to me for UI cleanup

openhardwarecoza, 2016-07-29:
[image]

prcdslnc13, 2016-09-05:
Is there any reason we have to run it as an A axis? Of we are generating the code internally and then driving from LW it seems a moot point. If your wanting to input code generated in CNC wrapper or something similar it should just be as simple as a find and replace post process code (or preprocess in the import )

You have to do something similar when 3D printing with MachineKit as they do not recognize E axis. So you have to do a pp script to replace E with A from a slicer.

openhardwarecoza, 2016-09-05:
(; because Wolfmanjm went through all the trouble to make us a proper abc
firmware (;

FearNaBoinne, 2016-10-03:
Don't want to be a party pooper, but not all ROUND items are CYLINDRICAL...  ;)

What about wine glasses, beer glasses, etc?

openhardwarecoza, 2016-10-23:
Moved to wishlist on the readme https://github.com/openhardwarecoza/LaserWeb3/commit/b39c56a24a7de76afdbf95dd3e28ae12f6bf6c8c so I can close the issue for now

closed 2016-10-23 by openhardwarecoza