LaserWeb/deprecated-LaserWeb3 #52: Next step in CAM UI:  Click entity and set operation

openhardwarecoza, 2016-07-27:
I already coded up the UI element, when we click on entities from a DXF:

[image]

You can select the individual entity, as of yet I am only adding a Bounding Box to it, and showing the size.  But the next level up might be to add a modal that pops up when you click it, with some cam options. 

This is to extend the layer concept a little.   

Right now each file becomes a layer, and for each layer you can set speeds, feeds/power and (in cnc mode) operation (inside, outside, etc) 

But it would actually be nice to click the entity and then set its cam operation/speed/power/etc

This would probably mean replacing this cam panel:
[image] with something that allows us a more tree like view of the File->its child entitities -> Operation created for that child

I am thinking https://www.jstree.com/ (I use it on my home automation app (https://github.com/openhardwarecoza/ESP8266-Sketches/tree/master/htt-websocket-esp8266) and it works well, plus after the previous project I already know the code.  

The caveat though, is to plan for the scenario where a user does not want to click a bunch of entities...

funinthefalls, 2016-07-27:
Would it be possible to enable it as an advanced feature, basic mode would leave it as it is?

openhardwarecoza, 2016-07-27:
I'd rather spend a little time making one feature, work well for both.  Maintaining two ways of doing one thing, becomes hell

funinthefalls, 2016-07-27:
I was thinking along the lines of it being more of a hide the feature in basic mode, show it in advanced mode way of doing it, not having two different sets of coding. Does that make more sense worded this way, or is it the same issue for you?

emteeoh, 2016-07-27:
jstree will track the usual values for the root, right? so until you select a sub-object and change its attributes, everything is root, and really not any different than how it is today, no? 

In other words, it sorta defaults to and degrades to what we already have, AFAICT. Seems like a good solution to me.

arthurwolf, 2016-07-31:
[image]

openhardwarecoza, 2016-07-31:
@arthurwolf (; did you just use a meme creator to make a point?  Lol!

openhardwarecoza, 2016-08-03:
Okay, so far I have opening files, populating the JStree:

[image]

Clicking on any parent or child in the tree,  places the bounding box on the correct object 

Seeing LINES in that sample user-submitted dxf - makes me realise for non laser toolpaths, its not going to be easy to handle lines...

diegotrap, 2016-08-03:
Maybe it would be interesting that the user is able to group some of those lines and define the same CAM parameters for them (feed speed, laser power, passes, etc). That way you avoid setting the same settings for many closed lines that in reality are the same entity for the user.

openhardwarecoza, 2016-08-03:
I aborted out of JsTree - just too much schlep for not enough features...  Couldnt easily extend it with options for the line

Now, I am using @Maxazan's http://maxazan.github.io/jquery-treegrid/ - which since its a table, allows me cells left and right of the tree leaves:  I need that for buttons:  

for example... Like:  (And yes this is live, go play with it)

[image]

emteeoh, 2016-08-03:
Looks great!
If you try to minimise the tree for a  file, it minimises all trees. Also, rather than delete lines/paths/etc, could they just be de-activated, so that they can be re-added if they were removed by mistake?

SigmazGFX, 2016-08-03:
Whoa, That's very similar to my undo inquiry.

openhardwarecoza, 2016-08-03:
@emteeoh true - just saw that too.  #bug

openhardwarecoza, 2016-08-04:
Added some more candy...

[image]

openhardwarecoza, 2016-08-04:
Another in-progress preview of the new CAM window:

1.  Opening DXF and SVG files populates the tree view with the file, and all its children entities.
2. Opening Bitmap files populates the tree view, but of course has no children. 
3. For each "file" you can move, adjust its dpi, delete, and select-all-children
4.  Alternatively if you are only interested in sub children of the object, you can select the relevant objects (many at once too) (clicking an entity in the viewer highlights its item in the tree as well so you can know what belongs to what)
5.  Once you have the geometry selected you want to work with, clicking Add Selection to Job,   copies that geometry to new threejs groups. And then adds those to the bottom panel (toolpaths) -  next we'll rewrite the Gcode generator to work from these Copies, instead of the original objects.  This gives the much requested feature of a) possibly exporting the job for reuse, and b) adding parts of a file to different jobs, give us the much requested capability of different feed/power within the same file as well
6.  Bitmaps are also added as toolpaths, so you can add/remove as relevant 
7.  I want to add two operations to bitmaps:  Raster,  and Vector Cut:  If you select vector cut, we'll transparently to the user, do a Potrace vector conversion and generate gcode from it
8.  Next to the jobs/toolpaths, you'll see delete and "Edit Parameters" buttons.  Edit is where you'll set speed/power for laser jobs,  and pocket/inside/outside/speed/feed for CNC jobs

[image]

openhardwarecoza, 2016-08-04:
The Resolution edit button on the SVG clicked, and the Move button on the DXF clicked:  Expands like this:

[image]

Note, functional: Already moves the items (:

arthurwolf, 2016-08-04:
I know I keep just spouting unhelpful praise all the time, but I really
wanted to say : this is awesome, keep it up.

openhardwarecoza, 2016-08-04:
Thanks Arthur!  Cambam like enough yet?  (;

arthurwolf, 2016-08-04:
> 5a6d-11e6-9e04-cd16d2b5aa82.PNG
> 237599164
> AAGpFSxOp2vCk6SuE4pKaRh6DcML6uz9ks5qcgyWgaJpZM4JV1la
> 237599740>,

Well, still got 327 features to implement to be cambam, but you are moving
in the right direction :)
Also in laser mode you probably don't want all of cambam. But there are
things that'd be nice to have ( a lot of which you had in v1 btw, so I'm
pretty sure you'll be putting them back in : resize, move etc )

openhardwarecoza, 2016-08-04:
Yip,  of course  (:   this time i am sort of going cnc first.  Easier to
simply for laser,   than it was trying to extend for cnc

openhardwarecoza, 2016-08-05:
Extended just a little bit (and a little bit buggy still) 

- but now, after creating a toolpath, you can click the 'eye and it hides everything, and shows just that "toolpath" or "job" 

 
[image]

funinthefalls, 2016-08-05:
Nice!!!!

kram242, 2016-08-06:
This just keeps getting better every time I see it! This new widget is going to be super helpful for adding complex toolpaths. _This is getting very pro looking Peter_ ;) **Awesome job**

emteeoh, 2016-08-06:
When you create 2 tool-paths, if you try to delete the second one, it deletes both!

openhardwarecoza, 2016-08-07:
Fixed, thanks @emteeoh

openhardwarecoza, 2016-08-07:
Making some more progress:  I didnt know you could use JQuery, to show/hide groups and elements, and update text inside SVG!   Now this changes, everything! (ideal thing to use for UI feedback whenever you need a graphical explanation)

[image]

arthurwolf, 2016-08-07:
That is *extremely* neat, this is totally the right direction to go into.
Also you need dialogs like this for way more things in laserweb :)

openhardwarecoza, 2016-08-07:
I hear you.  Modals are cool, but I also had too many modals in LW1 (;   

Is there anything in the current UI  that you think should move to a Modal?

arthurwolf, 2016-08-07:
The whole "connect" stuff at the top of the screen has no business being
displayed all the time : You use it once to connect, then it doesn't
matter. It should just be one button, that opens a modal. This also gives
you the opportunity to show/explain more things in the modal itself. Then
the button just displays the current connection status ( and reverts to
allowing you to connect again if disconnected ).

I really really wouldn't keep the "terminal" on-screen at all times. It's
useful to a dev and to somebody who is interrested in what is going on
behind the scenes, but to a "normal" user it's just screen-clutter. You can
have a button that opens it in a modal. Most laserweb user's ideal life
doesn't include manually writing/sending gcodes.
One thing the modal is useful for is displaying file upload progress. That
also needs to get it's own modal ( ideally just in the already existing
"upload" modal ). With a scrollbar.

With that cleared, you can also "overlay" the jog buttons on top of the
gcode preview, that way you get a larger gcode preview display.

I don't have laserweb installed on this machine. I have two dozen others of
those comments, want to get them all at some point ? :)

openhardwarecoza, 2016-08-07:
Well, I do have the file upload on modal already (;

[image]

"Terminal"  is way more than a terminal:  I print ALL the status updates to it; startup health check, feedback when saving and loading settings, error messages, feedback during file open, progress updates during gcode generation, etc - so actually very very useful and when I did have it hidden in on the first releases of LW3 - it led to chaos as no one went to open it and thus missed the feedback. 

[image]
Note: git messages, file feedback and a settings check all in there ^

Connect button;  I do agree - it can definately do with some rework. The current format is very much the "pronterface familiarity" thing ...

Overlay on the 3D viewer:  As this whole thread deals with advancing the cam to make objects in the 3D viewer clicable for cam paths,  this risk anything being overlayed - to become an obstruction to that action (and also confuses threejs's raycaster).  In LW1 we did a lot of overlay.   I agree it does look nice, but, per your argument, how often do you really jog? (;   If you machine has homing switches, I'd rather use those and offset positioning in the objects .  So if anything, i'd hide them even more. 

But yeah, open some extra issues - lets get all the parties talking.  I quite like this thread - feedback during dev is much less demotivating than before where I finished something, showed it off, then got shot down lol.

openhardwarecoza, 2016-08-07:
Re jog - moving them to the viewer doesnt actually give me back that horisontal space: Because:

Sidebar (; 
[image]

arthurwolf, 2016-08-07:
> Well, I do have the file upload on modal already (;
>
>
> I meant the upload progress. Last time I used laserweb that displayed in
the terminal.

> "Terminal" is way more than a terminal: I print ALL the status updates to
> it; startup health check, feedback when saving and loading settings, error
> messages, feedback during file open, progress [...]

I hear you. However, I taught a dozen people how to use laserweb a week
ago, they all would prefer that didn't exist ( or at least wasn't visible ).
There are definitely much better ways to say "Gcode generation is done"
than to display it in a terminal. Those users I mention didn't notice or
care about the terminal, and did miss the feedback anyway.

>
> Note: git messages, file feedback and a settings check all in there ^
Sure, again : the old lady that wanted to engrave a picture of her cat,
didn't care much for git messages :)

> Connect button; I do agree - it can definately do with some rework. The
> current format is very much the "pronterface familiarity" thing ...
>
> Overlay on the 3D viewer: As this whole thread deals with [...]
Oh I totally think jogging should be in a modal, I was just going one step
at a time :)

> (; If you machine has homing switches, I'd rather use those and offset
> positioning in the objects . So if anything, i'd hide them even more.

The users I mentionned did use jogging though : They place the plate to cut
at a random place on the work area, jog to it, set current position to
zero, start the job.

> But yeah, open some extra issues - lets get all the parties talking. I
> quite like this thread - feedback during dev is much less demotivating than
> before where I finished something, showed it off, then got shot down lol.

openhardwarecoza, 2016-08-07:
See theres a big difference between the "engineer" and "art" people... I would never be happy eyeballing where to laser (; - home + offset = repeatable.  lol

arthurwolf, 2016-08-07:
That's how a lot of people are going to be using laserweb though :)
That's what you get for making an easy to use interface : you get the kinds
of users who want easy to use interfaces :) Not engineers :)

openhardwarecoza, 2016-08-07:
[image]

Ok note to self: LaserWeb4 : 

[image]

openhardwarecoza, 2016-08-07:
I should stop watching Simpsons while coding...

funinthefalls, 2016-08-07:
You shouldn't type LaseWeb4 anywhere, someone will see it and expect you to have a working version done in 6 days ;) Anyways, it is only partly Laser now, need a great new name, maybe even get rid of the word web in the name?

openhardwarecoza, 2016-08-07:
Agreed.. Web can go!  (gets rid of all the xyz bad kharma too in the same
go)

funinthefalls, 2016-08-07:
Maker Studio,
CreateIt,
MotionHub,  
CSquared (Create and Control) Studio

Some tag lines could be:
"Computer Aided Making Tools"
"Intuitive Machine Control"

arthurwolf, 2016-08-07:
I actually quite like maker studio. It's missing an "Open" somewhere though.

funinthefalls, 2016-08-07:
Make the tagline "Open Source Computer aided making tools"

openhardwarecoza, 2016-08-07:
I cleaned up the form, and made it so it hides irrelevant fields based on tool, updates the values in the display above, etc.  Just a little more polish

animated gif again: 
[image]

openhardwarecoza, 2016-08-07:
Also added the dragknife diagram
[image]

SigmazGFX, 2016-08-07:
> 238071677>,

Did you say modal?

openhardwarecoza, 2016-08-07:
> auth/AAGpFaCW3oxvNmI4sWI-
> 238071921>,

Why?

openhardwarecoza, 2016-08-08:
Worked on the raster a little, and started adding the V-Bit peck feature to the CNC side:

[image]

openhardwarecoza, 2016-08-09:
FML: I hate SVGS!    I need help to get the SVG parser extended to give me the "layer" data :(  https://github.com/openhardwarecoza/LaserWeb3/issues/77

openhardwarecoza, 2016-08-09:
Also I got tired of scrolling down past all the object to work on toolpaths, so I split the panel into two "panels" each with their own sub scrollbar

[image]

cojarbi, 2016-08-09:
Maybe add a colapse entity also? After selecting I just need to work on the toolpath

AY

solarmill, 2016-08-11:
Would be nice to be able to engrave vectors (fill) instead of just cutting them.

cojarbi, 2016-08-11:
You can if you save it as a bitmap

AY

solarmill, 2016-08-11:
We have a professional workflow and all our artwork is in vector format.  We alternately cut or engrave/trace depending on the design.   It would be very odd for us to convert a vector into a bitmap just to fill it in.   This is a very basic feature of even the crappist chinese software, so it should be trivial to tell LaserWeb to either cut vectors or to fill the in.

cojarbi, 2016-08-11:
You still can, just save it as a different layer and adjust power.  I'm saying sour actual solutions now. 

There are other groundbreaking solutions in the works. 

Care to describe your actual workflow for reference? 

AY

solarmill, 2016-08-12:
Illustrator, VCarve Pro, Rhino, Fusion360....  When we create artwork, we do a single vector version and scale it to fit the size of the product we are marking it on.   Creating raster would require different versions for every product we own.   Also, we like to engrave AND trace the lines to get the best effect.   All our artwork files are .ai or .dxf.   We've been engraving (filling) with this files for years on a shitty chinese laser.   Was really hoping LaserWeb would be able to fill engrave them as well.

openhardwarecoza, 2016-08-12:
@SolarMill -  anything is possible,  but at some point i need to call in
help :
https://github.com/openhardwarecoza/LaserWeb3/wiki/Sponsors-and-Build-Team

openhardwarecoza, 2016-08-12:
> wiki/Sponsors-and-Build-Team

For what its worth.  I have already proven i can raster from svg.  But its
about 4-5hours of work to rewrite the svgloader to support the method...

cojarbi, 2016-08-12:
Is this the etche function in Corellaser? 

AY

openhardwarecoza, 2016-08-12:
Moved feature request to new issue.  Not related to this one (point and click and new cam interface)

cojarbi, 2016-08-12:
If I understand correctly you draw a filled vector and then raster that? 

AY

solarmill, 2016-08-12:
Yes, as long as the vector is "closed" the software should be able to fill it in (the same behavior as pocketing with a CNC machine).   You can't pocket an open vector, and you can't fill engrave an open vector

openhardwarecoza, 2016-08-12:
Gents,  move it to the newly created issue please (:

[9 more replies]

closed 2016-11-30 by openhardwarecoza