LaserWeb/deprecated-LaserWeb3 #54: WORKFLOW: Setting up a Multi-Step Laser Job

superdave42, 2016-07-27:
Simple workflow for setting up a multi-step job using LaserWeb.   In this example I use Illustrator,  but any vector authoring tool can be used.  This way you can run a single job that does multiple things. such as engraves/etches text, cuts through some internal shapes and then finally cuts out the outline shape.  All without having to setup 3 separate jobs to run on the work piece.

Dave

[Setting up MSLJ.docx](https://github.com/openhardwarecoza/LaserWeb3/files/386920/Setting.up.MSLJ.docx)
[Setting up MSLJ.pdf](https://github.com/openhardwarecoza/LaserWeb3/files/386921/Setting.up.MSLJ.pdf)

openhardwarecoza, 2016-07-27:
Man this is gold!   You make it look way simpler than I even planned it!
@funinthefalls any chance you want to help us get this into wiki asap (:

funinthefalls, 2016-07-27:
Am on it boss!

funinthefalls, 2016-07-27:
Will finish up both new workflows tonight, need to attend to some business first, but I really like thgese workflows. Great job SuperDave42!

superdave42, 2016-07-27:
I don't mind at all.  I plan to do some more as I get more playtime with LW.

funinthefalls, 2016-07-27:
Can I ask you to upload the images that you have in the docs, along with the text in a zip file in the future. Just makes it a little easier to edit and create the workflow wikis you are creating.

superdave42, 2016-07-27:
Absolutely.  for the 2 I created recently I just pasted my clipboard
directly into the Word doc as I went.  Next time I'll save the pics as
separate files and zip it all up.

funinthefalls, 2016-07-28:
Thanks SD

funinthefalls, 2016-07-31:
Yup, I will tackle these later today.

funinthefalls, 2016-07-31:
This one has been completed, simplified the photos somewhat to reduce the length, and edited content for grammar etc.

closed 2016-07-31 by funinthefalls

1stage, 2016-08-02:
This is pretty great. A couple of things I'd recommend:

- In Illustrator, turn on all layers and export a POSITION image file first. Import that file first into LaserWeb 3 as an aid in manually aligning the other layers. Until we have a clean import of Adobe Illustrator files with mixed vector and raster elements, using the offset to nudge the layers into place seems to be the only course.
- The workflow is pretty good, but it does assume a level of familiarity with LaserWeb 3 that not all might have, particularly in knowing where the buttons are (i.e. OPEN and PROCESS). Wouldn't hurt to create a specialized graphic in Photoshop that perhaps zooms in and identifies these items step-by-step.
- Although everyone's laser is different, showing how the layer parameters were set WOULD be of use to newbies. Perhaps take a little more time on that and don't jump to the finished product so quickly? You do a great job at the beginning (in the Illustrator section), but not so much at the end in LaserWeb 3.
- Finally, for simplicity, it might be easier to only have TWO layers (in addition to the previously suggested POSITION layer), because that's all that's really needed for this example: an outline cut and an etch pass. Having a CUT-THROUGH and a CUT-OUTLINE layer might make a newbie think this is required, when it's not.

I'm more than happy to assist in updating this workflow. I'm not up and running on SmoothieBoard and LaserWeb 3 yet (still burning-in the stock equipment), but I'd be happy to help.

superdave42, 2016-08-02:
Those are all good ideas /feedback  I like the position layer idea.