LaserWeb/deprecated-LaserWeb3 #55: WORKFLOW: Solidworks to export a Polyline .DXF file

superdave42, 2016-07-27:
Simple workflow for exporting polyline type .dxf files form Solidworks.  LaserWeb wants to see a polyline type file.  a Circle that is exported to dxf and is not a polyline type will cause the laser to start and stop as it goes around drawing individual lines to make up the circle.  This is not desirable.  We need to make sure our .DXF exports as polyline type.

[Solidworks to Polyline DXF.docx](https://github.com/openhardwarecoza/LaserWeb3/files/386987/Solidworks.to.Polyline.DXF.docx)

openhardwarecoza, 2016-07-27:
Also super!  CAM and Polylines go like bread and butter.    What caught my
attention there is that it can convert splines to polylines too...
@raykholo  -  (; solidworks as an expensive way to solve your spline
issues?

@funinthefalls -  heres another piece of gold!

raykholo, 2016-07-27:
That's awesome guys!  I do have solidworks...  ( @superdave42 if you aren't aware I was having the LaserWeb doesn't like splines issue from the DXF I was exporting form Autodesk Inventor).

openhardwarecoza, 2016-07-27:
@raykholo  shot in the dark,  but cant you customise dxf export in
Inventor?
https://knowledge.autodesk.com/support/inventor-products/learn-explore/caas/CloudHelp/cloudhelp/2014/ENU/Inventor/files/GUID-1EE5A959-8B4E-4924-8051-1E340A030B22-htm.html

raykholo, 2016-07-27:
@openhardwarecoza 

These are the options available.  I'm sure there may be some way to script it with the xml they mention, but from the GUI I have very limited options...

[image]

I'm actually quite happy with Vectric Aspire (more paid & not open source software, I know) - the Post Processor I wrote works, and now that I do everything as CAM (inner vs outer profile with a tool dia of 4 thou) I have my laser cutting with 2 thou dimensional accuracy.   I just use LW3 to send the gcode to the machine.

openhardwarecoza, 2016-07-27:
Traitor.    Post the xml as per the document i linked

funinthefalls, 2016-07-31:
Yup, I will tackle these later today.

funinthefalls, 2016-07-31:
Done

closed 2016-07-31 by funinthefalls