LaserWeb/deprecated-LaserWeb3 #56: SVG shapes are closed in Inkscape, open in LW3

emteeoh, 2016-07-27:
Sample attached, zipped because github doesn't let me attach a plain SVG. Actually, it seems to be closed everywhere *except* in LW3. 
[jet.svg.zip](https://github.com/openhardwarecoza/LaserWeb3/files/387128/jet.svg.zip)

openhardwarecoza, 2016-07-28:
Thanks, investigating...

cojarbi, 2016-07-31:
> I've got some new SVGs that are causing trouble... I finally sat down
> and started reading about the SVG format in the hopes of maybe
> understanding myself why the sample above was failing, and as an
> experiment, I broke the image up into 3 files. These new files render
> just fine in inkscape and the default image viewer in debian, but LW3
> can't load them at all. The error I see is "TypeError: Cannot read
> property 'length' of null
> (http://127.0.1.1:8000/lib/threejs/three.min.js on line 11)"... I'm not
> sure if I should open a new bug report for this, or just let it ride on
> this... I mean, I was investigating this bug, but the new error is more
> than likely at least a little bit my fault...
> [test
> svgs.zip](https://github.com/openhardwarecoza/LaserWeb3/files/392210/test.svgs.zip)
>
>
> ---

There has being some cases of this happening but still Peter doesn't know why, so this files will help. Nevertheless there is an option on settings to either open or close svg. 

AY

emteeoh, 2016-07-31:
Oh yeah, I know about the close svg feature... for reasons I forget, and I am sure are no longer relevant, I turned it off, but I don't need it off, so I have a work-around...

Looking at these though, it *really* looks like the "z" flag on paths is being ignored. I hesitate though... My reading is that z should cause an implicit line back to the start point, but the paths here specify no fill... I'm wondering if the threejs's coders are interpreting unclear portions of the SVG spec differently than mainstream...

cojarbi, 2016-07-31:
So this is what ive done.

Disabled auto close in LW3
Created a square in inkscape. By default it will show as a rectangle in layer 1. The i did the following:
Path > Object to path ( this will show as path with 4 nodes ). Render missing the left path
and also
Path > Stroke to path ( this will show as path with 10 nodes ).This last one will render correctly closer in LW3

Attached both files

openhardwarecoza, 2016-07-31:
Z is explicitly ignored yes,  since we want to place all paths flat on the bed to begin with,  then use the CAM to add 'Z'  moves away from 0. 

It does appear like we loose the first or last segment only.   But the caveat is,  it only happens for closed paths.  I.e things that have start and end of path supposed to join.     

But there are also use cases for open paths:   draw a few 100mm lines ,  imports fine.  Draw an arc,  it renders fine.   If we force autoclose on an arc,  it draws a line over the open face and it becomes a D shape.   

I've not looked at your files yet,  had some family things to attend to,  today as well.  But will try attend to it tonight  (:

emteeoh, 2016-07-31:
No rush! 

You're not talking about the same z I am. A simple closed path would have a "d" element that would look like this:
d="m 10,10 20,10 20,20 20,10 z"

the z in this case means "close this path", not the third dimension. If you're masking out z's without thought to context, maybe that's the problem?

openhardwarecoza, 2016-07-31:
Ahh ok,  no i meant when i draw I put everything at x=x,  y=y,  z=0.  Dont
know if svg even has a z height.  Dxf has.

You are saying something i didnt know though.   If the Z at end of path is
found,  we can autoclose the shape,  if not: dont.   So svg doesnt actually
add the last coordinate?   They just add the z and call it a day (is z then
meaning :  repeat first coordinate as last coordinate?)

openhardwarecoza, 2016-07-31:
And ps,  the svg import code is actually not from the threejs devs.   Its
my own modifications of the very old (2011) lasersaur svgreader script: i
added threejs draw functions on top of it.

Before this one,  we used snap.svg to parse the svg,  and that went even
worse!

emteeoh, 2016-07-31:
https://www.w3.org/TR/SVG11/paths.html#PathDataClosePathCommand

"So svg doesnt actually
add the last coordinate?   They just add the z and call it a day (is z then
meaning :  repeat first coordinate as last coordinate?)"

If you use Z or z in the path, yes. You can do an explicit line segment to close the path, but it handles line endcaps differently. That matters for rendering (and rasters, I guess), but probably not for cutting.

I'm still learning to make sense of javascript, otherwise I'd just give you a patch... found the svgreader code at least...

openhardwarecoza, 2016-07-31:
Awesome,  then yes,  thats the root of the bug!

openhardwarecoza, 2016-07-31:
The parser has a  Z
https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/lib/svgreader/SVGReader.js#L748-L758
so might be ib my svg2three code

openhardwarecoza, 2016-07-31:
Ok,  just reading over the code,  I dont see anything obvious.   Tomorrow,
i'll add some console.log statements inside that z case statement,  to see
if it eveb gets called.  If it does get called,  it should add that extra
segment in the paths.  And we seem to be processing all the paths (but it
wont hurt to console.log the parsed path count,  and our path count,  and
compare)

openhardwarecoza, 2016-08-02:
Ok, I spent some time tracing down the root failure point, turns out its not the parser: 

If I open a small rectangle (only 3 of the 4 edges show up in LW/Threejs):

The "paths" coming out of svgreader is indeed 5 vertexes (0,0 -> 0,100 -> 100,100 -> 100,0 -> 0,0)

In this code: (extra console.log lines I dont have on the commited version added): https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/lib/svgreader/svg2three.js#L246

the problem happens: 

```
    console.log("Shape.....");  //  Still shows 5 vertices
    console.log(svgShape)
    svgGeom = new THREE.ShapeGeometry( svgShape );
    console.log("Shape Geometry.....");  // Damn, now it only has 4 vertices!!! Why Three, why you throw away vertex
    console.log(svgGeom)
```

We take the "shape" which at the  second line still has 5 vertices,  as soon as I use the built in Three.Shapegeometry to create the usable "geometry" - drops one vertex!

Screenshot of the console log shows the above in action:

[image]

So at this point we can totally rule out the SVG Parser, even all the LaserWeb code:  Something in THREE.ShapeGeometry is causing us to loose a vertice:  - also seems to me looking at the paths extracted, it always drops the LAST vertex.   Shapes.autoclose works reasonable for shapes that should be closed, but some should not, so I really dont want to depend on shape.autoclose, if my path contains a vertice that closes it (;  - if three could stop ignoring it though (; 

[image]

Lets tag in the great ThreeJS @MrDoob and see if he can shed any clarity on why ShapeGeometry would do this...

openhardwarecoza, 2016-08-02:
[image]

openhardwarecoza, 2016-08-02:
@MrDoob - I reduced it down to a replicated JSFiddle:  https://jsfiddle.net/e7tbg5fp/3/

mrdoob, 2016-08-02:
If this is an issue with https://github.com/mrdoob/three.js, do you mind opening an issue there?

openhardwarecoza, 2016-08-02:
Thanks - doing to right away (:

openhardwarecoza, 2016-08-02:
[image]

Yay,so thanks to the threejs devs advice, (and apparently I was a dumbass for using ShapeGeometry)  its now fixed!

I will admit though, asking them for advice sucked.   Talk about being annoyed and not wanting to explain :/ 

Anywho, please test the new code (:

emteeoh, 2016-08-02:
Works! This fixes all my existing SVG errors, but it creates a new one. The attached file does something funny: the cut area disapears when I try to load it. Haven't looked too far into it yet. It worked yesterday, so I *think* its a regression...

[TieFighter final.svg.zip](https://github.com/openhardwarecoza/LaserWeb3/files/397366/TieFighter.final.svg.zip)

openhardwarecoza, 2016-08-02:
Checking in 5, just quickly grabbing dinner (:

openhardwarecoza, 2016-08-02:
Your file has a part some distance out to the right:

[image]

When we try zooming the extents on open, firstly that file is now a couple meters wide, and secondly,   when we zoom in the middle - we see a white window - since thats all there is between the left of the drawing and the right where that part is...

Delete that part then try again

closed 2016-08-02 by openhardwarecoza

emteeoh, 2016-08-02:
Ah! I wonder why that wasn't biting me last night? Musta moved it with a fat-fingered short-cut.

Ok then. It all works! Yay!