LaserWeb/deprecated-LaserWeb3 #57: WORKFLOW:  Use Macro Panel to create Laser Test-Fire button

superdave42, 2016-07-28:
Here is a WORKFLOW for creating a button that will pulse the laser for 1/10 second at a specified power level.    I wanted this for doing Mirror alignment  since I have removed the POT and test-fire button from my K40 control panel.  I was just using the button on the laser power supply but it blasts right through the tape leaving adhesive crap on the mirrors.   Now I can just use some thermal labels and just mark them by pressing the Maco button I created in the LaserWeb GUI.

[LaserWeb Macro Panel.docx](https://github.com/openhardwarecoza/LaserWeb3/files/387612/LaserWeb.Macro.Panel.docx)
G-Drive link with the images  ( would not let me post the ZIP here)  https://drive.google.com/open?id=0ByNecjb0RD-TbVdDMHdKVGV0WkU

funinthefalls, 2016-07-31:
Yup, I will tackle these later today.

funinthefalls, 2016-07-31:
Done, but will need some editing

funinthefalls, 2016-07-31:
Done

closed 2016-07-31 by funinthefalls