LaserWeb/deprecated-LaserWeb3 #61: Delayed coordinates display update when manual jogging.

Tegeon, 2016-07-29:
I found the coordinates showing X/Y/Z position are delayed (looks like machine 'Run' status is frozen for a little while) when manually jogging from 0,0,0 to X- or Y- or Z+ (jog by just one click using 10mm jogging distance or few quick subsequent clicks to multiply that distance). After few moments the displayed value gets to the right one and status goes to 'Idle'.

I am running default config.txt and no motors connected at the moment, latest firmware and latest git pull.

openhardwarecoza, 2016-07-29:
Usb or ethernet?

Tegeon, 2016-07-29:
USB at the moment. I need to solder WiFi or wait for Ethernet module delivery.

openhardwarecoza, 2016-07-29:
Ok,  will check in the morning.  Exact build of firmware you are on?

Tegeon, 2016-07-29:
Version currently Installed : 30188
Version available online on Github : 30188
Your version of LaserWeb is Up To Date!

openhardwarecoza, 2016-07-29:
(: I meant smoothie version,  so i test from same build.  Run 'version'  on
the command console. May need to catch output on the nodens if it doesnt
pop into browser log

Tegeon, 2016-07-29:
Here it is:
Smoothie
ok
version
Build version: master-27197b7, Build date: Jul  4 2016 04:10:42, MCU: LPC1769, System Clock: 120MHz

openhardwarecoza, 2016-10-23:
I havent been able to fix this just yet... I'm going to close the issue for now though, it boils down to a larger underlying thing on the serial buffer...

closed 2016-10-23 by openhardwarecoza