LaserWeb/deprecated-LaserWeb3 #67: Negative Zoom?

emteeoh, 2016-08-02:
If you keep zooming in on the origin, at some point the entire image flips and then starts to zoom out. Before and after flip screenshots attached.
[image]
[image]

openhardwarecoza, 2016-08-03:
Thats because when you zoom in past zero,  your camera now sits behind thw
grid.  And the camera will always face the user.  Thus,  to do that,  the
camera,  now behind the bed,  has to look back at you,  through he rear of
the bed.

Ie totally normal.  Just dont scroll that far.   I wont remove negative
camera positions since thats useful for inspection of the bottom of for
example cnc toolpaths (tools,  options,  cnc cam)

emteeoh, 2016-08-03:
It hadn't occurred to me that it was useful!

closed 2016-08-03 by emteeoh