LaserWeb/deprecated-LaserWeb3 #68: More SVG errors

emteeoh, 2016-08-03:
This time,I get "TypeError: Cannot read property 'length' of null (http://127.0.1.1:8000/lib/svgreader/SVGReader.js on line 681)"

The file is attached. I've already done the Object to Path in inkscape, but autoclose is gone, so I can't do that anymore. 
[F-117-1-cut.svg.zip](https://github.com/openhardwarecoza/LaserWeb3/files/399501/F-117-1-cut.svg.zip)

SigmazGFX, 2016-08-03:
It looks like your file is an Inkscape SVG.
I had some loading issues earlier today with one and the problem was resolved by changing the file type on save.
Could you try again, but this time saving a copy** of the file as "PLAIN SVG". 
(** not the original ! You'll lose the editors formatting)

emteeoh, 2016-08-03:
It is an inkscape SVG... But saving it as a plain SVG didn't help.

Looking at the SVG, I'm guessing the problem is in one of these elements:
 <flowRoot
     id="flowRoot4047"
     xml:space="preserve"
     style="font-size:10px;font-style:normal;font-variant:normal;font-weight:normal;font-stretch:normal;text-align:start;line-height:125%;letter-spacing:0px;word-spacing:0px;writing-mode:lr-tb;text-anchor:start;fill:#000000;fill-opacity:1;stroke:none;font-family:Sans;-inkscape-font-specification:Sans"><flowRegion
       id="flowRegion4049"><rect
         width="46.239407"
         height="43.379238"
         x="-248.83475"
         y="1734.6927"
         id="rect4051" /></flowRegion><flowPara
       id="flowPara4053" /></flowRoot>  <path
     d=""
     id="path4089"
     style="fill:#0000ff;fill-opacity:1;stroke:#0000ff;stroke-width:0.05958686;stroke-miterlimit:4;stroke-opacity:1;stroke-dasharray:none" />
  <path
     d=""
     id="path4097"
     style="fill:#0000ff;fill-opacity:1;stroke:#0000ff;stroke-width:0.05958686;stroke-miterlimit:4;stroke-opacity:1;stroke-dasharray:none" />
  <path
     d=""
     id="path4099"
     style="fill:#0000ff;fill-opacity:1;stroke:#0000ff;stroke-width:0.05958686;stroke-miterlimit:4;stroke-opacity:1;stroke-dasharray:none" />

Empty paths are weird enough, but I need to look up flowroot's. I wasn't aware I used such an object.

emteeoh, 2016-08-03:
OK, It wasn't the flowroot per se, it was the empty paths. I put "M 10 10" in them and it loaded fine.
Changing the name of the bug.

openhardwarecoza, 2016-08-03:
(: Can I log a bug report somewhere to find out where you guys find these nonstandard garbage files? An empty path?  Doesnt make sense! Why would there be a path... without content...

emteeoh, 2016-08-03:
Its an inkscape output. I *think* I know how I did it, but I agree, it makes no sense.

openhardwarecoza, 2016-08-03:
[image]

Also alerting the user - just in case every path is empty and they see nothing

closed 2016-08-03 by openhardwarecoza

emteeoh, 2016-08-03:
Nice!

I just a quick check and found two things of note:
- this was a bug in several mainstream browses at one point. You're in good company!
- the BNF actually allows an empty path! https://www.w3.org/TR/SVG/paths.html#PathDataBNF

openhardwarecoza, 2016-08-03:
See why i hate svg?

openhardwarecoza, 2016-08-03:
My push didnt go through.  Had a power failure. Just waiting for pc to boot back up,  then i'll push again

openhardwarecoza, 2016-08-03:
Done:  

https://github.com/openhardwarecoza/LaserWeb3/commit/a75e4228831307a4fc6b26698a9419528b46fabc

emteeoh, 2016-08-03:
From a parsing point of view, I still think SVG is better than DXF.  XML > hold_over_from_punch_cards