LaserWeb/deprecated-LaserWeb3 #69: Traced vector returns previous trace

SigmazGFX, 2016-08-03:
I think I may have read of this before but I can't find the issue.

Loaded bitmap A and traced to vector.
Deleted bitmap A
Deleted traced A layer

Browser not refreshed

Loaded bitmap B
Trace to vector returned vector trace from bitmap A

openhardwarecoza, 2016-08-03:
Confirmed - reproducable with those steps...

openhardwarecoza, 2016-08-03:
Found why:  On removing the tab, i delete all the threejs objects, but I dont delete the origImage elements from the dom:  See fiverr.png (my first file) still present, with same ID

[image]

openhardwarecoza, 2016-08-03:
Done:  

Added ...

```
if (objectsInScene[layerIndex].type == "Mesh") {
       $("#origImage"+seq).remove();
 }
```

closed 2016-08-03 by openhardwarecoza