LaserWeb/deprecated-LaserWeb3 #71: Optimise raster for machines with M3/M5

openhardwarecoza, 2016-08-03:
M3/m5 machines currently get fed a m3/m5 for each segment of the raster.  Its already optimised in the vector code (if laser is not already on... https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/js/threegcode.js#L261) 

Same can be done for Raster G1 X moves:  https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/js/laserraster.js#L256  (add some isLaserOn?  (laseron here refers to do we have a laseron command to add or not.  needs another if statement around it)

pflugj, 2016-08-04:
I tested with the JTECH patched Version 0.9g.

This Version does not except the "; " as comment and I get an error at line 1: "Expected command Letter" 
So I saved the G-Code and removed the comment lines. 
I then used GRBL-Panel to run the G-Code and the lasing starts as expected.

But there are still some lines where the M3/M5 statement is present.
This happens whenever a "S0" is present in the "G1...   " command lines.
So I edited the file, removing all M3/M5 except the once at the beginning and the end and it works. 
 ...no more stopping and the resulting picture is much better ;-)

When using the same feedrate for black & white it also seems to be unnecessary to send the feed-rate with every command.
Therefore I then removed all "F....." statements from the G1 command lines,
except for the very 1st one at the beginning of the file. "G1 F......"

I then even removed all "G1 " from the command lines,
except for the 1st one leaving just "X..." and "S..." in the command lines
.....and it still works ;-)

Surprisingly the patched version of GRBL does change the feedrate while moving too.
... at least that is what it sounds and looks like.

[image]

openhardwarecoza, 2016-08-04:
I cannot (and wont)  make changes that work for "grbl only"  -
Smoothieware is the favourite here.  And anything the gcode generators
create has to work on ALL platforms.   I am still waiting on anyone else to
complain about the comments.    If anyone else reports it too,  i'll strip
them

pflugj, 2016-08-04:
Is smoothie not GRBL compatible when it comes to the commands?

pflugj, 2016-08-04:
I would love to use smoothie but my C-Beam machine does have NEMA23 steppers and as far as I understand Smoothieboard can only supply up to 2A per stepper, am I right?

openhardwarecoza, 2016-08-04:
no, just plug in external drivers into the step/dir pins,,,,

openhardwarecoza, 2016-08-04:
smoothie is grbl compatible but supports way more - so its not bothered by things like ;

Anyway, check the grbl documentation (official, not some jtech shit fork) and see if it doesnt support ;  - as I said in that best practice thread... NO GUESSING - bring me facts

openhardwarecoza, 2016-08-04:
Added some code, git pull and test...

openhardwarecoza, 2016-08-04:
More detailed replies:  (you like writing a lot)

I tested with the JTECH patched Version 0.9g.

```But there are still some lines where the M3/M5 statement is present.```
    - yes since laser is NOT always on... 

```So I edited the file, removing all M3/M5 except the once at the beginning and the end and it works. 
...no more stopping and the resulting picture is much better ;-)```
Looks good indeed - but needs cross platform testing. 

``` When using the same feedrate for black & white it also seems to be unnecessary to send the feed-rate with every command. ```
I actually don't know one person that doesnt need the adaptive feedrate.... You should try it... 

```I then even removed all "G1 " from the command lines  ```
Yeah but really not worth it - grbl is not that SLOW that that extra 2 bytes is worth the effort - it helps keep the gcode readable - which to me means more

``` Surprisingly the patched version of GRBL does change the feedrate while moving too. ``` 
Exactly, so stop living in the past.  0.9c is soooo old

openhardwarecoza, 2016-08-04:
Another topic on the S0 - remember the bigger picture too:  On smoothie we have whats called "tickle power" so S0 may be 10-20% on a tube machine...   You really dont want that on during a G0 move...

pflugj, 2016-08-04:
-- More detailed replies: (you like writing a lot)
I am just try to avoid further misunderstandings like yesterday ;-)

--yes since laser is NOT always on...
I know, but "S0" is spindle off and M5 is not needed in this case, am I right?

-- Yeah but really not worth it - grbl is not that SLOW that that extra 2 bytes is worth the effort - it helps keep the gcode readable - which to me means more
I thought that would give you the extra bytes which btw. are 3 bytes including the space ;-)

I just did a git pull and thecomment lines starting with ";:  are still in

btw it is JTECH GRBL v0.9g and the unpatched v1.0c does not complain about the ";"

openhardwarecoza, 2016-08-04:
No M05 = spindle OFF,   S0 = spindle speed to 0... Not exactly the same:   Again, remember:  a world outside Grbl:  Both Marlin,  and Dual pin Smoothies (https://github.com/openhardwarecoza/LaserWeb3/wiki/Firmware:-Smoothie-Dual-Pin-Config-(PWM-and-FIRE-pins) have TWO pins:  One controlled by S, and one controller by M03/05....  - so no, not the same

It adds the comments (i am not throwing them away) but look where the commit is:  https://github.com/openhardwarecoza/LaserWeb3/commit/86cd1465a8e8b6b7481d9243810f83e82450f86e in server-grbl === ie NOT in the frontend:  its in the backend, on the serial bridge - it strips out the lines WHILE sending over serial

openhardwarecoza, 2016-08-04:
3 bytes at 115200 bps = 0.0015625seconds... If your laser needs to move that fast, you shouldnt be using 8 bit....

pflugj, 2016-08-04:
=== ie NOT in the frontend: its in the backend,
Oh, ....I will test again later

openhardwarecoza, 2016-08-04:
I try and avoid "firmware specific"  but if I absolutely have to, i'll stick it in server-<firmware>.js -   i dont want to maintain three frontends (the backend is only 280 lines long... The entire git project is currently at 96530 lines (run wc -l $(git ls-files) in the repo directory on linux)  -  so yeah, i'd rather maintain specifics in the 280 lines file than anywhere over 96,000 lines!

openhardwarecoza, 2016-08-16:
@pflugj  - i'm going to close this issue for now.  But later when you have more time, would love to have you work on finding more optimisations! that last einstein looks awesome!

closed 2016-08-16 by openhardwarecoza