LaserWeb/deprecated-LaserWeb3 #72: POWER CONTROL  S1 - S255 ?    I'm used to S0.1  to S1.0  for the GCODE

superdave42, 2016-08-04:
I thought the power control is from 0 to 1   and that 80% would be 0.8
now I see when I generate Gcode it is putting the PWM value at the end of S

is this normal ?     for a test I put 20% power for the laser and the resulting GCODE comes out as S51  instead of S0.2   

Just seeing if this is normal behavior   because S51 makes my laser run full blast.

I am using the latest version of LW3

openhardwarecoza, 2016-08-04:
Set max pwm value to 1 in settings -  gcode.    Syslog should be telling
you that a)  you forgot to configure it and b)  while rastering the syslog
will say 'using default value of 100 since max pwm isnt configured'
(seems nobody reads the syslog.  Should i switch to popup notifications?)

cojarbi, 2016-08-04:
If it's required then pop up

AY

openhardwarecoza, 2016-08-04:
Yah well, if you configured it AS 255, thats a valid config (grbl and marlin uses it...) so I cant really flag on that...   I flag if its EMPTY and print:

[image]

Not easy to autodetect that sort of user induced misconfiguration. 

Moral of the story: Users need to know the gcode requirements of their controller (;   - the documentation team could help fill the wiki with more help files though :(

superdave42, 2016-08-04:
Totally understand,  Thank you for getting me back on track.   User error,

cojarbi, 2016-08-04:
Let's do a table with the commands per controller

AY

openhardwarecoza, 2016-08-04:
yip, and like I added help link on the settings -> tool window: each setting needs that.  Our documentation team is however otherwise occupied

superdave42, 2016-08-04:
Here is a graphic if it helps at all
[image]

superdave42, 2016-08-04:
Also the default   says  255 in that window until you actually set it,

openhardwarecoza, 2016-08-04:
Its not filled in.    Its a 'placeholder'  -  note how its not black but
gray?   Its just random content put in there (to avoid people asking 'how
must i type'  :/ you can never win.   Consider these niggles a basic
intelligence test (;   if you cant figure out how to get your gcode setup,
maybe you shouldn't play with lasers lol

superdave42, 2016-08-04:
lol fair enough..

SigmazGFX, 2016-08-04:
Hah!!

openhardwarecoza, 2016-08-04:
Unless I dumb down all the sample placeholder fields to something like this?

[image]

superdave42, 2016-08-04:
probably just something that can be added to the Wiki instead  when bandwidth is available to add.  I can make screen shots etc  to help.   Now that I know it, I won't forget it.   My GCODE knowledge is getting better every day.

SigmazGFX, 2016-08-04:
I too thought the placeholder was a suggestion that could be used if you didn't know... like the pre and post gcode boxes have a legit homing code.
one option is to change the placeholder to a valid possible value.( ? )

cojarbi, 2016-08-04:
Yes but there are codes specific for the platform. 

AY

openhardwarecoza, 2016-08-04:
Well, since we had three users today who got confused be the placeholders, I am reopening and renaming...

The way I see it theres a few solutions:

1.  Change the placeholders from an "example" to a description "Please type here this that yada"  

I hate doing that! Users hate it when you treat them like idiots too (even if they are)

2.  Change the CSS of the Placeholder to an even lighter gray - if you google "bootstrap placeholder too dark" you'll see its not just us that have this issue with default bootstrap theme

3.  Replace the "config" tab with an onboarding wizard (ever powered on a new iphone and had to spent 10 mins on that startup wizard joining 3 different icloud services, and telling it your name and date of loss of virginity, etc etc before you could use your device?   We could do that, but i fear that would discourage users who just "want to see what it looks like" if we force a full config procedure before the UI loads.

4.  Do 3, but with a skip or "do later" button... Yeah... uhm, right now you start with an empty config and the syslog tells you to configure it.  Users dont go and configure it.  So I have a feeling asking them to do it "later" will result in "never"

5.  Auto sane settings - we tried it in LW1, didnt work so well, and made code maintenance hell

cojarbi, 2016-08-04:
I like #2, could pre configured options for common setting per controller be a solution? To bypass #3 situation. 

That's no without adding the relevant documentation teaching / explaining the user

AY

openhardwarecoza, 2016-08-04:
Here's what it looks like with a much lighter placeholder

[image]

openhardwarecoza, 2016-08-04:
@cojarbi  - good point, maybe just a one pager popup asking "smoothie or grbl?"

superdave42, 2016-08-04:
I like the lighter placeholder,    but it's not possible for everyone to understand that's just a placeholder,    so it still needs to be beefed up with documentation at the end of the day I guess.

the pop seems cool, intuitive.

SigmazGFX, 2016-08-04:
That's what I was about to post.
It's not the placeholder that is the issue. it's the users understanding of what goes in there.
That can be rectified in the Wiki

cojarbi, 2016-08-04:
Yes

AY

openhardwarecoza, 2016-08-04:
I find it shocking that people dont understand placeholders.  Every single login on the web these days forgo the "label" and put the placeholder in the login.  Well not all, but 99% of the services I use

superdave42, 2016-08-04:
Well I thought I was pretty smart to.  oh well.    just posting so you have
a feel on what may be causing confusion for others.  trying to help

openhardwarecoza, 2016-08-04:
no man, as I say, you were the 3rd person to confuse the placeholder with a value... so its a real problem.  Just not sure how best to solve (users find a way of complaining about any plan lol)

openhardwarecoza, 2016-08-04:
Well make it 4th: https://plus.google.com/106033318557602563181/posts/dvCRF7h6Bag

SigmazGFX, 2016-08-04:
> dvCRF7h6Bag

Wow..
It really is a thing.

openhardwarecoza, 2016-08-04:
> 237686173>,

Yeah,  and thats been that way for months. No changes,  everyone just
managed on their own...  Which of you started this prank (;

openhardwarecoza, 2016-08-04:
Oh and apparently LW3 is also a non communicative unclear interface.
Sigh.  Remind me again why we do this?

SigmazGFX, 2016-08-04:
> 106033318557602563181/posts/
> 237695937>,

I'd like to point out that I was merely agreeing that at first glance I was
slightly unsure.
I don't think I ever complained, but I was using a ramps setup then so I
had plenty of other things to fight with.

And you're doing this for the chance at world domination.

openhardwarecoza, 2016-08-04:
> issues/72#issuecomment-
> 237696735>,

Lol.   Oh yes!  That reason.  So,  not masochistic... Lol.

SigmazGFX, 2016-08-04:
> auth/AEAhHFEVygZ8XHtUw05p_
> 237697923>,

Haha. Nah, masochism is so last month.
One things for sure, the more I complain....err, ask the more I learn.
No better way to become profficiant than to harass the dev.

superdave42, 2016-08-04:
Whatever reason you are doing this,  It's the most fun I've had with DIY so
far.   LW  blows the pants off using Repetier or pronterface to run GCODES
that came out of some inkscape extention  Uggh..
it's clear you spend a LOT of time on this project.   I hated using my K40
for so long..   now it's my favorite maker thing.
so THANK YOU for all your hardwork and for not giving up on this.

openhardwarecoza, 2016-08-04:
> 237698366>,

(; the secret lies in the way you do that (; -  with most of you i look
forward to being harrased.  I actually like teaching.

  If i dont,  you get banned (; so if you can still see my replies,  i like
you.  Lol.

emteeoh, 2016-08-04:
Woah! I made the cut?!?!

I don't think the LW3 UI is at all unclear! its not an easy subject, but it does an OK job of it...

funinthefalls, 2016-08-05:
I will expand on the documentation this weekend. Will start with the basics of setting up LaserWeb for the first time, with specific examples for each controller.

funinthefalls, 2016-08-05:
You do this for the tens of dollars you will make after 1000's of hours of coding ;)

funinthefalls, 2016-08-09:
Upgraded documentation for this  I-D-1-0-T  issue is almost complete.

closed 2016-08-16 by openhardwarecoza