LaserWeb/deprecated-LaserWeb3 #73: Spot size resets on browser refresh if 0.1

cojarbi, 2016-08-05:
Select 0.1 as spot size
Refresh browser
It will ask you again for the spot size

emteeoh, 2016-08-05:
I agree this is a bug, but the work around is to enter something like 0.11. Or 0.101 if you think half-a-thou is too much error.

cojarbi, 2016-08-05:
Ohhh I remember, wast it you that reported it. I was traveling and did not make sense to me at the moment

AY

emteeoh, 2016-08-05:
Wasn't me who reported it... But I read it.

funinthefalls, 2016-08-05:
I set mine to 0.1000001

openhardwarecoza, 2016-08-06:
the fact is, as I say in that warning, that rastering actually delivers better results when the spot is slightly defocusses (note almost like Feathering in photoshop) - so a 0.1mm spot size is actually quite unlikely

I only have around  of the 240 active users on the new version, so theres still serveral users on the old version.  Can't remove the "please change it" popup until some more people have upgraded.

cojarbi, 2016-08-06:
Agree. Just note that I was testing on a diode and 0.5 on cardboard looked like thick lines

AY

openhardwarecoza, 2016-08-06:
> 238016513

Thats a science we really need the hive mind to tackle.  Not just finding
whats best,  but also how else we can optimise...  The way @pflugj tested
here :
https://github.com/openhardwarecoza/LaserWeb3/issues/71#issuecomment-237503203
(though my response came off irritable because of that other thing that
happened,  sorry @pflugj)  -  its actually awesome work to manually
manipulate the gcode and see the results.   Really made me realise we can
do even better rasters!  But what need more people like @pflugj  playing
with this,  find solutions that work gcode wise,  and i'll code it into the
code...

closed 2016-08-09 by openhardwarecoza