LaserWeb/deprecated-LaserWeb3 #74: Raster generation is much slower than before.

funinthefalls, 2016-08-06:
Using the same file, it used to take about 10 seconds, now it is taking 3 1/2 minutes.

funinthefalls, 2016-08-06:
From the sys log:

: [Rasterizer] 99% done

: [Rasterizer] 99% done

: [Rasterizer] 99% done

: [Rasterizer] 99% done

: [Rasterizer] 99% done

: [Rasterizer] 99% done

: [Rasterizer] 99% done

: [Rasterizer] 99% done

: [Rasterizer] 99% done

: [Rasterizer] 99% done

: [Rasterizer] 99% done

: [Rasterizer] 100% done

: [Rasterizer] 100% done

: [Rasterizer] 100% done

: [Rasterizer] 100% done

: [Rasterizer] 100% done

: [Rasterizer] 100% done

: [Rasterizer] 100% done

: [Rasterizer] 100% done

: [Rasterizer] 100% done

Raster completed in 58311ms

: Raster Completed for butterfly-digital-stamp.png

: Preparing Gcode File: butterfly-digital-stamp.png

: Estimated Distance: 148.4 m

All sorts of duplicate % completed entries.

openhardwarecoza, 2016-08-06:
Remember,  we now change the resolution,  to match spot size.  Look at the
comments in laserraster.js with the math,  but basically we have a lot more
data to process when we really run 0.1mm lines.   Typically before if you
took a 72dpi image you had what?  A line every 25.4/72 = 0.3x mm?

funinthefalls, 2016-08-06:
Ok,I understand, forgot about that.

closed 2016-08-06 by funinthefalls