LaserWeb/deprecated-LaserWeb3 #76: image import top left

fumblz-n-action, 2016-08-09:
if you import an image into the top left corner it still imports the image to 0,0 when it comes to generating Gcode. the image is visually in the top left corner of the UI but  Y offsets aren't automatically compensated for. moving the y in negative offset results in gcode being generated outside the work environment and will result in machine crashes for people not using min and max endstops on both x,y

funinthefalls, 2016-08-09:
This is more a "User has not configured his machine correctly" than a software issue. In fact LaserWeb is behaving exactly as it should (If you are like me it will only take Peter 7-8 attempts at explaining why it is correct before the light bulb over your head snaps on.

openhardwarecoza, 2016-08-09:
[image]

I tested just now and generates gcode at the top:  First draws a nice Y400 G0 move before starting the  path

Can you elaborate with screenshots?

openhardwarecoza, 2016-08-09:
Unless I misread your post, "negative offsets" - if you tell it to put the file outside the  envelope and home, prior to running, well of course you'll crash.  But think of the usecase of users without endstops (banggood lasers) - where you might use the SET ZERO - in which case you may consciously have negative offsets for some reason (so i also dont want to limit the x/yoffset inputs to positive numbers)

fumblz-n-action, 2016-08-09:
it happens when you move the image with a -Y offset
[image]

fumblz-n-action, 2016-08-09:
sorry i misworded it last night. i was getting pretty tired

openhardwarecoza, 2016-08-09:
Had a + instead of a Minus on https://github.com/openhardwarecoza/LaserWeb3/commit/b8103a1469cbba7abc7706acbb850b3eb69e6912#diff-c8ca0d5f58d3f66fb668f05549c2745bL191

openhardwarecoza, 2016-08-16:
I assume you would have commented if the fix (please test above) didnt work.  So closing this one (:

closed 2016-08-16 by openhardwarecoza