LaserWeb/deprecated-LaserWeb3 #77: HELP WANTED:   Add Layer parsing to SVGReader + SVG2Three

openhardwarecoza, 2016-08-09:
For DXF I have an awesome parser that gets me the Layer attribute for each polyline/line/circle/etc 

[image]

I would LOVE to have the same for SVG (especially going forward with #52 ) -  but I need some help adding it to 

https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/lib/svgreader/svg2three.js  

and

https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/lib/svgreader/SVGReader.js

openhardwarecoza, 2016-08-10:
@lautr3k :  inkscape left,  illustrator right: Looks like id= the common field we can use between files

[image]

thats from Yuusuf's files I tagged you in on G+

lautr3k, 2016-08-10:
almost done...
[image]

openhardwarecoza, 2016-08-10:
Haha!  Yeah sorry,  slow me was still thinking parser and here you are
already a million lightyears ahead!  This is sokk awesome!

lautr3k, 2016-08-14:
Done!

closed 2016-08-14 by lautr3k