LaserWeb/deprecated-LaserWeb3 #79: Can't change macro button colour.

emteeoh, 2016-08-10:
I've got a few macros, and when I go to edit, I can select a colour for the macro, but after saving, the macro is back to the original colour. It looks like the only time I can successfully change the colour is when creating a new macro.

emteeoh, 2016-08-10:
Hmm. No, not quite: I can also change the colour of the *LAST* macro.

emteeoh, 2016-08-11:
Looking through code to see the problem, I wonder about https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/js/macros.js#L113
It starts at 1. Shouldn't it start at 0?

openhardwarecoza, 2016-08-11:
> blob/master/public/js/macros.js#L113

I think row 0 of the table was the header?  (: my plane just landed in
Johannesburg,  disembarking now.   (sorry,  off to customers so cant look
till Tonight)

emteeoh, 2016-08-11:
Yeah, I think you're right. When I look at the readmacros, you also start at 1, so you're at least consistent.

openhardwarecoza, 2016-08-11:
Lets drill down,  see ifs the save or the load: look at dev tools -  local
resources,  macro1,  change its color and save.  Check devtools again
before refresh: did it update?

openhardwarecoza, 2016-08-11:
Inspect the element too:
https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/js/macros.js#L131
-  that color+i input,  does the color picker change its value?

emteeoh, 2016-08-11:
I'm looking at this for the first macro:
<span class="btn-colorselector" style="background-color: rgb(50, 205, 50);"></span>
and this:
element.style {
    background-color: rgb(255, 140, 0);
}
When I click on a colour, they both update correctly.

I'm not sure how to check the element after a save "before the update". Sounds like you want a JS "breakpoint", but I have no idea how to set that.

openhardwarecoza, 2016-08-11:
Ahh yes,  thats true,  this updates live not after a page refresh.   Hmm.
Ok,  will check tonight.

emteeoh, 2016-08-11:
Hmm. Watching the JS console, I see that when I select a colour, something updates a macro I don't have:
Setting button  9
macros.js:161 Setting Color  #colorValue9 #FFD700 #FFD700 gold
macros.js:161 Setting Color  #colorValue9 #FF8C00 #FF8C00 darkorange
macros.js:161 Setting Color  #colorValue9 #FFD700 #FFD700 gold
macros.js:161 Setting Color  #colorValue9 #6495ED #6495ED cornflowerblue
macros.js:161 Setting Color  #colorValue9 #32CD32 #32CD32 limegreen

I have 8 macros, and I was changing the colour of macro1, not 8, let alone 9. How that all translates into me being able to change the colour of 8 but nothing else... that's beyond me at the moment.

emteeoh, 2016-08-11:
The callback at 159... when its called, how does i get initialized?

emteeoh, 2016-08-11:
So confused. More playing, and I realized that changing the colour of any macro changes the colour of the last macro, but the error doesn't show until some time after you click save...

emteeoh, 2016-08-11:
hmm. Breakpoints! So, the call back gets three parameters: value, color, title. The debugger shows that the values passed in are: title = "#DC143C", color = "#DC143C", title = "crimson". Seems like you're missing an element name somewhere.

openhardwarecoza, 2016-08-11:
Okay, lunch break check here:

Catch problem:  On chromium, under Ubuntu 16:04, and I cant replicate the issue.  I create a macro,  edit again, change the color, save, shows new color :/

openhardwarecoza, 2016-08-11:
Line 159:  https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/lib/bootstrap/js/bootstrap-colorselector.js

openhardwarecoza, 2016-08-11:
only now seeing your other responses...

openhardwarecoza, 2016-08-11:
Setting button  1
macros.js:161 Setting Color  #colorValue1 #6495ED #6495ED cornflowerblue
macros.js:154 Setting button  2
macros.js:67 Setting Color  2 #DC143C #DC143C crimson
macros.js:154 Setting button  3
macros.js:67 Setting Color  3 #DC143C #DC143C crimson
macros.js:154 Setting button  4
macros.js:67 Setting Color  4 #FF8C00 #FF8C00 darkorange

Setting color (154-161)gives me a different resonse than yours to-o

openhardwarecoza, 2016-08-11:
value = returned from the plugin when its clicked

emteeoh, 2016-08-11:
I did a video capture of what I see... https://youtu.be/HLuOUONKK74

openhardwarecoza, 2016-08-16:
Back from Johannesburg, tested on Chrome 52.0.2 on Win10: Still not able to replicate the issue...

Are you sure you dont have any funny characters in the macro name or gcode that could not be escaping correctly?

openhardwarecoza, 2016-08-16:
Closing issue, but do continue commenting... I still get notified.

closed 2016-08-16 by openhardwarecoza

emteeoh, 2016-08-16:
I'm not *sure* but I don't think so.
[laserweb-settings-backup.json.zip](https://github.com/openhardwarecoza/LaserWeb3/files/419712/laserweb-settings-backup.json.zip)

I'm wondering if its an extension like lastpass...

emteeoh, 2016-08-16:
I get the same behaviour on firefox.