LaserWeb/deprecated-LaserWeb3 #80: Opening multiple times the same file name wont process

cojarbi, 2016-08-10:
I was testing the signal (PWM) sent to the laser with a V display and opened the same file multiple times to change the power setting in each and then generate a consecutive run. 

It stays as the image bellow. Only processes the first. 

So either same name file updates the current one or we process them all ( if possible ).

Tested with dxf and svg

[image]

cojarbi, 2016-08-11:
Further testing shows that the processing is not working if you just change the name of the file. You need to edit the file ( ai , inkscape ) and change the size itself for it to work

cojarbi, 2016-08-11:
Attaching the files

[Archive.zip](https://github.com/openhardwarecoza/LaserWeb3/files/412368/Archive.zip)

emteeoh, 2016-08-11:
Are you sure its not processing? I saw this a few weeks ago, but ultimately concluded it had generated g code for all the instances of the file, but mistakenly put all the g-code in the first image box... Not sure why I didn't file a bug...

openhardwarecoza, 2016-08-11:
I wouldnt stress this too much,  the new cam is coming along so quickly and
that portion is being rewritten anyway

closed 2016-08-11 by openhardwarecoza