LaserWeb/deprecated-LaserWeb3 #83: Raster from vector files

openhardwarecoza, 2016-08-12:
As requested by @SolarMill and others.   I have proof of concept code, but its a few hours of work to rewrite several functions to support it.  Beerfund contribution can help me free some hours off work to do it github.com/openhardwarecoza/LaserWeb3/wiki - > Sponsors

cojarbi, 2016-08-12:
Beer fund beer fund beer fund

AY

solarmill, 2016-08-12:
I just contributed to @openhardwarecoza's beer fund.   Does @cojarbi have a beer fund?

cojarbi, 2016-08-12:
Double up to @openhardwarecoza 

AY

solarmill, 2016-08-12:
Done 👍

openhardwarecoza, 2016-08-13:
In progress screenshot:

[image]

[x] Loading up the SVG file loads it as both a vector and raster
[x] In the new CAM (#52) you have the option of adding either the raster or the vector or both to the final toolpath
[ ]  Actually generating gcode - not yet (: but should be done by end of the week if my schedule stays flexible enough

openhardwarecoza, 2016-08-13:
Lets take it a step further:

Quick screen2gif capture,  showing all layers, toggling between vector and raster view while showing availble toolpath options (raster options on the top object, vector options on the bottom object) 

GIF Animation:
[image]

solarmill, 2016-08-15:
This looks great.  Will it be able to engrave fill .ai and .dxf?

openhardwarecoza, 2016-08-15:
Probably not.  We dont support AI at all (Proprietary format).   Dxf,  hard
to do,  no fill data.  Only vector line

solarmill, 2016-08-15:
Correct me if I'm wrong, but you can you process .dxf to cut and pocket in the CAM setup, right?

openhardwarecoza, 2016-08-15:
The cnc operations are clipper offset paths (concentric)  not raster.  Not
the same code at all.  Not even close (;

openhardwarecoza, 2016-08-16:
@lautr3k  - stuck on a issue here thats also an ideal time for me to gain some more experience on the scoped vars we were talking about on hangouts:

In bitmap rasters we have Bae64 image data to work from when I do : https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/js/laserraster.js#L154-L156

For SVG, raw SVG XML is not an accepted input for the paperjs new Raster(). 
For that, I need a canvas (so the familiar new Image from the SVG data, then draw canvas from the img during an onload, and finally then call the new Raster. 

```
var img = new Image();
img.onload = function() {
    ctx.drawImage(img, 0, 0);
    // From the onload we need to call the  this.raster = new Raster({ source: <idofnewcanvas) }); in order for paperjs to read the svg
}
img.src = object.userData.imgdata;
```

The complication is that paperjs does not seem to read the SVG directly unless you first put it into a canvas.   Loading the image to draw the canvas happens asyncronously, so I need to use the img.onload. 
And then the part that trips me:   Inside the onload, thisraster is not the "this" thats outside the onload... "this" is the img....    
So my question (teach me master) is how do I scope then the this.raster var correctly?

I was thinking of moving the https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/js/laserraster.js#L154-L159 into a new function and calling that from the onload?

openhardwarecoza, 2016-08-16:
Reference to paperjs:raster:source http://paperjs.org/reference/raster/#source

lautr3k, 2016-08-16:
`var img = new Image();
var self = this;
img.onload = function() {
    ctx.drawImage(img, 0, 0);
    self.raster = new Raster({ source: <_idofnewcanvas_) });
}
img.src = object.userData.imgdata;`

openhardwarecoza, 2016-08-16:
Awesome! Trying it that now...

openhardwarecoza, 2016-08-16:
Done in https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/js/laserraster.js#L154-L184

Okay, so,  now that this.raster gets generated, but the found function doesnt get executed:

Refer to my testing on https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/js/laserraster.js#L172-L173 -  If I comment out L172, and uncomment L173,  I get the console.log("Event Fires!")) event.   If I comment L173, and uncomment L172:  I dont get any error, but I also don't get the console.log at the top of the onRasterLoaded function I am binding to: https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/js/laserraster.js#L342

It of course works fine for the not-an-svg part here https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/js/laserraster.js#L176-L184

matt-chaney, 2016-08-21:
For what it's worth, AI hasn't been proprietary for several years. They're just open standards compliant PDF files.

solarmill, 2016-08-21:
@matt-chaney, exactly!   It's also one of the most used vector formats in the design industry.   SVG is growing, but still a teeny tiny fraction of the pie.    Support for the most common vector formats (.dxf and .ai) is an important feature.  Also, there's no need for the artwork to be defined as "filled" or not.  If the vectors from closed, non overlapping loops, they should get engraved.  If they are open lines, they get ignored.

solarmill, 2016-08-21:
The following GPL programs are capable of reading and modifying .ai files (from wikipedia):

**Ghostscript** (GPL-licensed program) can convert AI postscript files into various formats, and also allows programs that use Ghostscript such as GIMP to import AI files.
**Inkscape** (a free, GPL-licensed program) can import the old PostScript-based Adobe Illustrator format (up to AI ver.8) and AI files based on PDF (AI 9.0 and above).[9][10]
**Scribus** (a free, GPL-licensed Desktop publishing program) can open, edit and embed both PostScript- and PDF-based AI files.
**Skencil** (GPL-licensed program) also claims to have some AI support.
sK1 (a free, GPL-licensed vector graphics editor) supports PostScript-based AI files up to AI ver. 9.[11]

Perhaps their code can be a guide.

openhardwarecoza, 2016-08-23:
Find an AI to three.js library (;

openhardwarecoza, 2016-09-27:
Back from being offtopic,  i am still stuck on the call back : 

@lautr3k Done in https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/js/laserraster.js#L154-L184

Okay, so, now that this.raster gets generated, but the bound function doesnt get executed:

Refer to my testing on https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/js/laserraster.js#L172-L173 - If I comment out L172, and uncomment L173, I get the console.log("Event Fires!")) event. If I comment L173, and uncomment L172: I dont get any error, but I also don't get the console.log at the top of the onRasterLoaded function I am binding to: https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/js/laserraster.js#L342
(nor does the function continue. We definately arent reaching it)

It of course works fine for the not-an-svg part here https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/js/laserraster.js#L176-L184

openhardwarecoza, 2016-10-23:
Nudge - 

@lautr3k Done in https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/js/laserraster.js#L154-L184

Okay, so, now that this.raster gets generated, but the bound function doesnt get executed:

Refer to my testing on https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/js/laserraster.js#L172-L173 - If I comment out L172, and uncomment L173, I get the console.log("Event Fires!")) event. If I comment L173, and uncomment L172: I dont get any error, but I also don't get the console.log at the top of the onRasterLoaded function I am binding to: https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/js/laserraster.js#L342
(nor does the function continue. We definately arent reaching it)

It of course works fine for the not-an-svg part here https://github.com/openhardwarecoza/LaserWeb3/blob/master/public/js/laserraster.js#L176-L184

openhardwarecoza, 2016-10-28:
And here we have it - an SVG rastered! @DarklyLabs and @solarmill  - please test

[image]

I'll still refine the UI a little - at the moment when you open an SVG it loads two tabs. One's the SVG, one's the "raster" of the SVG - so the UI is clunky - you got to close the tab you dont want... I literally just got the callback working so I pushed that so long (before I break it) 

I'll work on making the UI friendlier tonight

funinthefalls, 2016-10-28:
Works OK for me. Nice job @openhardwarecoza

iceman1979, 2016-10-29:
Do all SVG files from Inkscape contain raster data even if there is no filled areas and it is all paths?
[image]

openhardwarecoza, 2016-10-29:
Read the comments in
https://plus.google.com/+PetervanderWalt/posts/bTQ3KsSHfSj
Still in progress.  For now close the tab you dont want

openhardwarecoza, 2016-10-29:
[image]

Nevermind that i literally also explained it 3cm above the reply button (facepalm)

iceman1979, 2016-10-29:
Ah I see, I misread what you wrote.  I went back and re-read what you wrote and yeah, I read it wrong. Well, I read it right, but my brain interpreted it wrong lol. I'm not even going to say what I was thinking LOL. WOW that was stupid LOL.

openhardwarecoza, 2016-11-27:
Work on this is now continuing in #178

closed 2016-11-27 by openhardwarecoza