LaserWeb/deprecated-LaserWeb3 #84: Add default svg dpi resolution to config

cojarbi, 2016-08-12:
If as user mostly uses ai or inskcape they can set a default dpi and thus not need to change at every load 
[image]

openhardwarecoza, 2016-08-13:
Could work... but... Even better:  Remember the new work on SVGReader? 

We seem to be able to determine from the header meta data if its is Inkscape, Illustrator or something else (: -  Just waiting for Sebastien to finish the SVG Parser work - once thats merged,  I'll add in this automatic detection (:

: Note: See how the new parsers contains an "inkscape" object:
So, I can simply check for "if (svg.inkscape) { var dpi = 90; }" (;
[image]

cojarbi, 2016-08-13:
Oh man, how could i forget the black magic sebastien is cooking

lautr3k, 2016-08-15:
Done !
[image]

closed 2016-08-15 by lautr3k