LaserWeb/deprecated-LaserWeb3 #86: Image movement after gcode generation

fumblz-n-action, 2016-08-13:
if you complete a job and want to run again by offsetting or changing the size of the object. once code image is moved the existing gcode preview stays on screen. this is only cosmetic as it does not run the code

openhardwarecoza, 2016-08-13:
Cool I picked up on it the other day, but wasn't sure at which step i got it to happen... Seems move is what breaks it.  Will investigate...

cojarbi, 2016-08-13:
indeed its move / offset
- Load and generate raster
- do a position offset
- generate raster again
[image]

openhardwarecoza, 2016-08-16:
Pushing a fix: Please test

cojarbi, 2016-08-16:
I get the same result.
[image]

openhardwarecoza, 2016-08-17:
Found the flaw in my logic. Fixed now

cojarbi, 2016-08-17:
comfirmed

closed 2016-08-17 by openhardwarecoza