LaserWeb/deprecated-LaserWeb3 #89: viewer.js error message

funinthefalls, 2016-08-14:
TypeError: Cannot read property 'userData' of null (http://localhost:8000/js/viewer.js on line 786)

Not surer what this error is?

openhardwarecoza, 2016-08-14:
Dont act like a user now... where, when what?  when does it error?

lautr3k, 2016-08-14:
Probably by loading a mesh...

funinthefalls, 2016-08-14:
Sorry, was going back over it so I could step by step it for you but you beat me to the punch.
Load bitmap.
Set DPI to 72
All other settings left untouched
Generate Gcode. 
Generates Gcode, but then spits out the error message in the trerminal. Not sure what it is a out, the Gcode looks fine.

funinthefalls, 2016-08-14:
I checked with 4 different bitmaps, all 4 throw the same error after generating the gcode.

closed 2016-08-14 by lautr3k

funinthefalls, 2016-08-14:
Thanks, error is gone. That was some fast fixing guys!

commit 30f75bd8 (2016-08-14): Fix #89 + indentation