LaserWeb/deprecated-LaserWeb3 #91: Padding for raster run-up acceleration on X dir changes

openhardwarecoza, 2016-08-15:
@fumblz-n-action 's issue:  New Edge build of smoothieware (and all the lesser firmwares) needs a 2-3mm padding to run up to speed on direction changes for raster.  (Also recommended by @arthurwolf)

I need to add a 2-3mm margin on laserraster.js

openhardwarecoza, 2016-09-02:
Smoothie team found the Bug in the edge branch! We dont need to change anything! Yay!

closed 2016-09-02 by openhardwarecoza

arthurwolf, 2016-09-02:
You totally need to change stuff :) Just not for this specific issue :) The
varying gcode length thing is severely limiting quality on smoothie-based
boards. Hopefully skarab merges his new stuff soon :)

openhardwarecoza, 2016-09-02:
Its exactly because of the things he's busy with that makes me close this...