LaserWeb/deprecated-LaserWeb3 #93: Tree-Based CAM toolpath preview bad DPI

emteeoh, 2016-08-15:
Steps to reproduce:
load a 90dpi image
go to TBC
create a toolpath
configure toolpath
select preview toolpath
==> the toolpath is at 25.4DPI, but the SVG is at 90.

openhardwarecoza, 2016-08-15:
https://plus.google.com/+S%C3%A9bastienMischler-Skarab/posts/WcbDEKyS6gW related.   Sort of duplicate of #94

emteeoh, 2016-08-15:
I don't think its a duplicate of 94. In 94, the image is loaded correctly, but the DPI is incorrectly displayed but only initially. Here in 93, everywhere I look, the DPI says 90, but the preview toolpath seems to be ignoring the 90 using 25.4 instead. Even if I edit the DPI of an image and explicitly set it to 90, the toolpath still incorrectly renders.

lautr3k, 2016-08-15:
Not the same but steadily related.

openhardwarecoza, 2016-10-06:
Is this issue still an issue?

closed 2016-10-23 by openhardwarecoza