LaserWeb/deprecated-LaserWeb3 #94: Single File load DPI inconsistent

emteeoh, 2016-08-15:
I load an SVG created in Inkscape. It correctly loads, and is displayed at 90DPI, but the image pulldown in CAM shows it as being 25.4DPI. 
[image]

If I change the DPI to 90, the rendering remains unchanged, confirming it was loaded at 90DPI.
[image]

If I now change the DPI to 25.4, the rendering changes to what I expect 25.4 to look like. 
[image]

openhardwarecoza, 2016-08-15:
Relates to https://plus.google.com/+S%C3%A9bastienMischler-Skarab/posts/WcbDEKyS6gW  

Agreed though, the "default dpi" should be set to that file's input field... @lautr3k

lautr3k, 2016-08-15:
I look at it tomorrow, need investigation... not so simple, if you reload the same file again it's work, or if you load multiple file too.

emteeoh, 2016-08-15:
Just confirming:

If I load 1 file, DPI displays wrong.
If I load multiple files in a single operation, DPI is correct.
If I load multiple files, one at a time, DPI for the most recent file is wrong, but all the others are correct, even if they were incorrect before the latest file load.
If I load multiple files then one or more single files, the multiple files are correct, but the newest single file is incorrect.

emteeoh, 2016-08-15:
I should point out that I set my default DPI to 90 (as well as for inkscape, 96 for adobe), so the 25.4 is coming from something hard-coded, not my config.
[image]

lautr3k, 2016-08-15:
Thanks for the recap :)

cojarbi, 2016-08-15:
I can confirm and replicate. 

Will add that if I delete the files via the tabs, the last preview stays on the screen even when the tab is gone. This is with svg

Mind that if one of the files has an embedded raster It will show the symptom above. 

Not sure if it's related, let me know and I open another case

AY

emteeoh, 2016-08-15:
Ok, this is fixed, but the tool-path generation is not.  Should we leave this one open, re-open 93, or make a new one? :)

openhardwarecoza, 2016-08-15:
This is the hard part of 'need to work on upcoming features,  and not break
existing features'  lol

lautr3k, 2016-08-15:
We need a dev branch.

openhardwarecoza, 2016-08-15:
I know we do.  But look how much and how quickly we get feedback when break
live things (;   -  I count that a win (;

lautr3k, 2016-08-15:
Reopened #93

closed 2016-08-15 by lautr3k

cojarbi, 2016-08-15:
Ohhh the feeling of being a lab mice, but this lab is amazing!!!  :) 

AY

emteeoh, 2016-08-15:
If you had a dev branch, I'd probably be running it and giving you feedback anyhow. I'm trying these features *BECAUSE* you said they're there, not because they're in main.

commit e4f63e1c (2016-08-15): Fix #94