LaserWeb/deprecated-LaserWeb3 #95: Add default dpi for raster

cojarbi, 2016-08-16:
A user most surely will use a common dpi when exporting raster. Same as vector add a default option to config and leave the option to be changed after import.

krolco, 2016-09-01:
I'm working on this. I have most of it, except the initial position of the scaled raster object.

openhardwarecoza, 2016-09-01:
Thanks @krolco !

krolco, 2016-09-02:
I just submitted #104 for this.

openhardwarecoza, 2016-09-02:
Merged.

@cojarbi you can test

cojarbi, 2016-09-02:
Works as advertised, Thanks

cojarbi, 2016-09-07:
@krolco Had some issues today, raster will not show at all. That was because in another machine i did not set the default required, I suggest we place a default value that can be overwriten in the cam tab

krolco, 2016-09-07:
I caused another problem with this ( #114 ) so I have to think about it a bit more. Grr.

krolco, 2016-09-08:
Hi @cojarbi, I am looking at #114 a bit. I'm not sure what you asked about on this one?
You mentioned that it didn't raster because you didn't fill out the required default. Were you suggesting that there should be a default value when the app is first started?

krolco, 2016-09-08:
OK, I added a quick default to the CAM tab -- if the defaultBitmapDPI isn't set, then it defaults to 25.4 DPI for an imported image. I'll get that pulled in just a moment.

krolco, 2016-09-08:
OK, I merged this in #119

cojarbi, 2016-09-08:
ok its good now

cojarbi, 2016-09-08:
@krolco after further inspections there is another issue.
Whether you place a value or not in the settings, that value does not get passed to the CAM section so no gcode is generated. 

It is taken into consideration while importing the image and i did not went through the generating process until now

krolco, 2016-09-09:
Well that's just strange. Let's reopen this one and I will try to figure this one out.

krolco, 2016-09-10:
@cojarbi I think I found my newest bug here, and again it's related to another fix :)  When I was fixing the problem for  #95 and #114, I moved the dpi/scale definition to another section of the code, but the CAM section was still using the (now undefined) value. I found that and passed it through (correctly) from the current object scale. I'll try to get this merged up later today.

krolco, 2016-09-10:
Ok, I merged #129 for this... I think this should finally be fixed! Sorry for the trouble.

cojarbi, 2016-09-10:
Confimed, thanks

krolco, 2016-09-10:
Thank you! I'm going to close this one again, with garlic and a stake...

closed 2016-09-10 by krolco