LaserWeb/deprecated-LaserWeb3 #96: Add default jog speed

cojarbi, 2016-08-17:
Can we have default jog speed in config also?

openhardwarecoza, 2016-08-17:
Instead of default in settings - I think "keeping last used value" may make more sense?

cojarbi, 2016-08-17:
I like that.  Is just a repetitive task that I don't think you need to move very often 

AY

lautr3k, 2016-08-17:
A step further we could think of a profile system ?

openhardwarecoza, 2016-08-17:
Eventually yes.. Pronterface currently uses the "last used" way and its quite nice

krolco, 2016-09-02:
This seems like it's in my wheelhouse. I'll try to swing at this weekend.

openhardwarecoza, 2016-09-02:
Note @wolfmanjm just added laser power override to smoothieware too :)  so
getting queue jump to work means we can override feed and power (;

krolco, 2016-09-02:
@openhardwarecoza I think this is a different issue?

openhardwarecoza, 2016-09-02:
Oh shoot.  Yes.  Sorry!  (:

openhardwarecoza, 2016-09-02:
(; or maybe i was subconsciously hinting at priority hehe

krolco, 2016-09-02:
LOL :)

krolco, 2016-09-07:
Pull request is in for this -- @cojarbi, please give it a test after it's merge.

cojarbi, 2016-09-07:
Thanks @krolco.  BTW can we start a beer  round for you? What's your PayPal.me?

krolco, 2016-09-07:
LOL, not yet. :) Let me fix my introduced bugs first.

cojarbi, 2016-09-07:
Set it up and send it to me please, our group supports every single
developer.

cojarbi, 2016-09-08:
Is there anything in the GUI that will show this has been saved? I changed my jog speed, refresh but it doesnt stick or is it something esle?

krolco, 2016-09-09:
Let me try it out. I saw it working on my setup, but maybe something else isn't working.

krolco, 2016-09-09:
Ah, I think I didn't understand the request -- I made LW3 save the *jog distance*, (0.1/1/10/100mm) setting, and remember that between sessions, but not the jog speed. I can add that also.

cojarbi, 2016-09-09:
Ohhh. Yes the speed gets reset on every browser refresh 

AY

krolco, 2016-09-09:
OK, I'll add that also. Sorry about the confusion!

krolco, 2016-09-09:
I have this implemented with a "Save Jog speeds" button locally like this: https://1drv.ms/i/s!AhNJtMTn2rnyh3-A5Gt6hoJy3huM

Tell me what you think, and I can commit this.

krolco, 2016-09-09:
I have an open pull request ( #127 ) that should implement this, if you like the save button.

cojarbi, 2016-09-09:
It look good and fits. I imagine that Peters suggestion to just grab the last value on the field and use it is not possible? My original suggestion was to set it up on settings with the rest of them, thus manteining the same logic for the user. 

Let's thinks for a minute, if I'm using one machine I'm most likely won't be changing the jog speed after I find the apropiate one. But what happens if I'm connecting to a laser and a CNC then the value could be the same or not?

Then again future machine profiles could solve this matter. 

AY

krolco, 2016-09-09:
I could save the last value in the field; however it's less when you are done typing in a number; other parts of the app work that way, though, so I can change it to that system instead of a save button if you and Peter prefer. Let me know.

krolco, 2016-09-09:
I meant "less clear". Other parts of the app a triggered on keyup, which means saving as you type. I'm not a fan of that, but that's just a preference.

cojarbi, 2016-09-09:
I'm in favor of placing a default field in settings and then the user can change it as pleased. Make sense for me much more if we are getting profiles. 

Let's wait for Peters opinion here

AY

krolco, 2016-09-09:
There is a default: it will default to 30/5 if you haven't ever changed settings; I left those defaults in. The difference is whether or not there is a save button, or whether we automatically try to save the setting when somebody types into the field.

I am testing a change right now that triggers off the Javascript "change" functionality: you can type a number in, but the "change" event isn't fired until the field loses focus, so the settings is saved when you either tab or click off the field. 

There are a few other parts of the app that trigger off the "key up" event (like settings the offsets for raster and meshes); I'm not a huge fan of that, because it can cause conversion errors (for instance, if you're trying to type "300.0" and you've typed "300." that can fail text->number conversion. That's why I prefer the "change" trigger event, but it can sometimes causes unexpected behavior for users since it isn't fired until the field loses focus.

And I tried to skip *all* of that by just making it an explicit user action to click the save button. :) I am going to commit a version that uses "change". I think that it will act the same way Pronterface does (which was referenced in the original request for this item).

M

krolco, 2016-09-09:
OK, I committed this without the save button, with automatic saving on change triggers. I'm going to merge it in this state, and it's just a little merge-fu if Peter decides he wants it the other way.
I'm also going to close the other pull request.

krolco, 2016-09-09:
#128 should do this now. Please give it a spin.

closed 2016-09-10 by cojarbi