LaserWeb/deprecated-LaserWeb3 #98: Enhancement: Object placement

fumblz-n-action, 2016-08-26:
is there a way to add the function to grab objects to move around the work fields using the mouse? don't get me wrong moving with a physical offset is more than adequate. the only reason i notice this is i have added objects in the wrong order... Example : cutting before raster... i have to remove the cut path and then re-add it. a +- layer button could solve setting up the offsets once again as well... i also notice if a layer is removed from the mix it removes any of the proportional feeds and cutting speed data along with power min max settings the user defines on all the other objects left but will still maintain the defined resolution of a raster image

cojarbi, 2016-08-26:
Managing the layers, processes etc will be handled by the new tree option where you process individually what you need in the order that you need. 

I do se a reset on the settings when deleting LW layers

AY

openhardwarecoza, 2016-08-26:
We had https://plus.google.com/+ArielYahni/posts/6xe5sLssshu in April for dragging objects around but it was really clunky and didnt work well.  Came from http://threejs.org/examples/misc_controls_transform.html

Happy to accept pull requests for something better... 

Layers: As said, the Tree function is in progress and will solve that: Follow progress on #52

openhardwarecoza, 2016-09-02:
Closing this one for now... Lets revisit in LW4

closed 2016-09-02 by openhardwarecoza