LaserWeb/deprecated-LaserWeb3 #99: Make LaserWeb NPM Installable

openhardwarecoza, 2016-08-26:
I published LaserWeb onto NPM:  https://www.npmjs.com/package/laserweb

but when I try installing it it installs the dependencies, but doesnt grab the LaserWeb files off git?

[image]

Out of my depths here, but hey getting this working makes the LaserWeb install a 2 step process:

1) Install node
2) npm install laserweb

RoaddogLabs, 2016-08-28:
It works.  The module is installed in the node_modules dir.  I don't have a Smoothie powered laser here (mine is an FSL) but it appears to be functional

[image]

cojarbi, 2016-08-28:
Indeed i can confirm that npm install laserweb does work. Mine was located at /Users/ayahni/node_modules/laserweb/ when executing tha install from the user profile root

cojarbi, 2016-08-28:
Well that was premature. I does load and shows available devices but while trying to connect terminal display :
Missing error handler on `socket`.
TypeError: SerialPort is not a function
    at Socket.<anonymous> (/Users/ayahni/node_modules/laserweb/server-smoothie.js:180:14)
    at emitOne (events.js:77:13)
    at Socket.emit (events.js:169:7)
    at Socket.onevent (/Users/ayahni/node_modules/laserweb/node_modules/socket.io/lib/socket.js:335:8)
    at Socket.onpacket (/Users/ayahni/node_modules/laserweb/node_modules/socket.io/lib/socket.js:295:12)
    at Client.ondecoded (/Users/ayahni/node_modules/laserweb/node_modules/socket.io/lib/client.js:193:14)
    at Decoder.Emitter.emit (/Users/ayahni/node_modules/laserweb/node_modules/socket.io/node_modules/socket.io-parser/node_modules/component-emitter/index.js:134:20)
    at Decoder.add (/Users/ayahni/node_modules/laserweb/node_modules/socket.io/node_modules/socket.io-parser/index.js:247:12)
    at Client.ondata (/Users/ayahni/node_modules/laserweb/node_modules/socket.io/lib/client.js:175:18)
    at emitOne (events.js:77:13)

cojarbi, 2016-08-28:
So doing np, install on that folder fixes and connect. Seems that serialport does not get installed

RoaddogLabs, 2016-08-28:
What platform did you build for?  I built against the macOS Yosemite 10.11.5 and Debian Jesse in Parallels on that machine.  Both load as expected.

I couldn't install the npm tools or node using Brew on the latest macOS Sierra dev beta (dev 7).  This system has compliers and dev tools from Xcode beta 8 though there are some updates available I haven't yet installed.

cojarbi, 2016-08-28:
Im on El Capitan 10.11.3

RoaddogLabs, 2016-08-29:
Sierra dev7, node 6.4, npm 3.10.3, modules auto downloaded, displays as expected.  There is an enoent error but I haven't looked into it to see what was missing.

I've got a [branch](https://github.com/RoaddogLabs/LaserWeb3/tree/electron-app) using [Electron](http://electron.atom.io/) to package as a standalone app.   Slack, the Atom editor and Wordpress are a few of the companies using it.  It allows for porting or development of a server side node app to a standalone desktop app (Mac/Win/Linux) that is packaged and install as any other desktop app.  It's going to require some non trivial rework.  For example it no longer needs a web server and the serial module has some issues.

[image]

openhardwarecoza, 2016-08-29:
awesomeness. i investigated electron before but the packaging of
node-serialport was the hard one.  that was back in node 0.12 days
(laserweb1 days) though. i hope it improved along the way :)    part of why
in Laserweb2 we tried SerialPortJSONServer (bigish mistake) and back to
node for lw3.   the node backend works amazingly well. but i do agree
installation to users could be friendlier.

RoaddogLabs, 2016-08-29:
The npm package still needs some Win testing but it looks ready for public consumption.  The node/npm install is pretty easy then running npm install laserweb.  It still requires the terminal but I think most may find that more palatable than a full build from scratch.

As a reference, the current primary issue with the Electron branch as follows...

The first run in the screen cap is with the serial module set to install.  The program will load after an error but will crash when the serial button is clicked.  I've rebuilt the modules both from npm and from the electron module rebuild tool.  I'll have to dive into it and see exactly what and where using the serial module is doing since the dump doesn't mention the serial module specifically but the error disappears when I no longer call the module.

The second run is with the serial module commented out.  The app opens with no errors, no crashes. 

[image]

cojarbi, 2016-08-30:
I can test in win, what's the procedure to follow?

RoaddogLabs, 2016-08-30:
I tested with Win7 under emulation on a Mac (it was good), here's what I did... (I made it a basic overview so we can cut and paste this to the Wiki if the masters of docs approve...)

1) Download node from https://nodejs.org/  I use the current version though the LTS version will work as well.  Once finished the download appears as a .msi installer.  Run it and you'll have node and npm.

2) Open the command prompt (terminal).
  * start menu  
  * all programs
  * accessories
  * command prompt

3) The terminal opens in `C:\Users\<YourUserName>`.  Remember in Win/DOS speak the `ls` command is `dir` and `/` is `\` .

4) Check your node/npm install
  * Run `node -v && npm -v`  which should return the respective version numbers.  If not, something happened in your node install.

5) Install LaserWeb3 and the dependencies 
  * Run `npm install laserweb`  It will create a `node_modules` dir if you don't already have one.

6) Change dir to the laserweb dir
  * Run `cd node_modules\laserweb`

7) Start LaserWeb3
  * Run `node server-smoothie.js`  and LaserWeb3 should start.

funinthefalls, 2016-08-30:
Nice job, I will add to the wiki once Peter approves!

funinthefalls, 2016-09-01:
Added to the Wiki. Thank you RoaddoLabs for getting this done for us. :)

closed 2016-09-01 by funinthefalls